carla-simulator/carla #2954: Importing Traffic Signal from RoadRunner

Mkarami3, 2020-06-15:
Hello Everyone,
First, I wanted to thank CARLA team for providing an exciting project for free.

I created a simple Junction with Traffic Signal in RoadRunner and using Export->CARLA in RoadRunner. I created the files ready to be imported into UE4. 
The problem I have is that when I push "Play" butting in UE4, each traffic signal is shown as three different signal. Did anyone have this issue before.
Your helps would be much appreciated.
Screenshot from before pushing "play" button, which is similar to what I created in RoadRunner:
[image]

ScreenShot after pushing "play" button: you see two new traffic lights are created for unknown reason
[image]

closed 2020-06-18 by Mkarami3

HarrisDePerceptron, 2021-01-06:
Hi @Mkarami3. Good to see you solved this duplicate traffic light simulation issue.  We been having the same issue.  we did scene rotation from tools> transform scene > Rotation and change that field value to 180 but it did not solve the problem.  Can you elaborate a little more what exactly you did. did you rotate each individual traffic light 180 degree along z-axis ?
thanks in advance

Mkarami3, 2021-01-06:
> Hi @Mkarami3. Good to hear you solved this duplicate traffic light simulation issue. We been having the same issue. We did scene rotation from tools> transform scene > Rotation and change that field value to 180 but it did not solve the problem. Can you elaborate a little more what exactly you did. did you rotate each individual traffic light 180 degree along z-axis ?
> thanks in advance
> 
> Road Runner: R2020b
> Carla (build from source)

Hi Harris,

If I remember correctly, I did rotation in RoadRunner 2020 but I still had the same issue. So, I used a different version of CARLA.
I am not quite sure which version, but I think it was CARLA 0.96.
Hope that helps