carla-simulator/carla #2955: The Installation of CARLA using apt has problem

atinfinity, 2020-06-15:
## Summary
I read the following manual.

- https://carla.readthedocs.io/en/latest/start_quickstart/#carla-installation
- https://carla.readthedocs.io/en/latest/ros_installation/

And, I tried to install `carla-simulator` and  `carla-ros-bridge-melodic` on Ubuntu 18.04.

### carla-simulator
```shell
$ sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-keys 92635A407F7A020C
$ sudo add-apt-repository "deb [arch=amd64 trusted=yes] http://dist.carla.org/carla-0.9.9/ all main"
$ sudo apt-get update
$ sudo apt-get install carla-simulator
```
### carla-ros-bridge-melodic
```shell
$ sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-keys 81061A1A042F527D
$ sudo add-apt-repository "deb [arch=amd64 trusted=yes] http://dist.carla.org/carla-ros-bridge-melodic/ bionic main"
$ sudo apt-get update
$ sudo apt install carla-ros-bridge-melodic
```

But, I found some problems.

## Problem
```shell
$ cd /opt/carla-simulator/bin
$ ./CarlaUE4.sh -windowed -ResX=160 -ResY=120
```

```shell
$ roslaunch carla_ros_bridge carla_ros_bridge_with_example_ego_vehicle.launch
```

But, I occured <https://github.com/carla-simulator/carla/issues/1716>.
I think that `numpy` and `pygame` are old version.

So, I installed the latest version via pip.
As a result,  I was able to avoid the problem.

```shell
$ sudo pip install numpy==1.16.6
$ sudo pip install pygame==2.0.0.dev10
```

atinfinity, 2020-06-15:
And, I tried to use carla_waypoint_publisher.
But, I could not launch carla_waypoint_publisher.

```shell
$ roslaunch carla_waypoint_publisher carla_waypoint_publisher.launch 
Traceback (most recent call last):
  File "/opt/carla-ros-bridge/melodic/lib/carla_waypoint_publisher/carla_waypoint_publisher.py", line 196, in find_ego_vehicle_actor
    self.reroute()
  File "/opt/carla-ros-bridge/melodic/lib/carla_waypoint_publisher/carla_waypoint_publisher.py", line 168, in reroute
    self.current_route = self.calculate_route(self.goal)
  File "/opt/carla-ros-bridge/melodic/lib/carla_waypoint_publisher/carla_waypoint_publisher.py", line 217, in calculate_route
    dao = GlobalRoutePlannerDAO(self.world.get_map())
TypeError: __init__() takes exactly 3 arguments (2 given)
```

I think that `carla-ros-bridge-melodic` is too old(0.9.7-1).

joel-mb, 2020-06-23:
Hello @atinfinity ,
Sorry for my late response and thank you for opening the issue. You are right, the debian package for the `carla-ros-bridge-melodic` is too old and crash with the current version of carla. I'll update it during this week.

Thanks!

germanros1987, 2020-06-25:
@joel-mb please, report back once this is solved and close the issue.

joel-mb, 2020-07-02:
Hi @atinfinity 
Sorry for my late answer. We are working on fixing this. We need to update the `carla-msgs` package in the ROS index before updating the ros bridge package, so this may take a few days. I'll keep you informed.

Thanks!

joel-mb, 2020-08-31:
Hi @atinfinity,
All the debian packages have been finally updated, so the issues should have disappeared. The installation process has been modified a bit. Now, both packages (i.e., `carla-simulator` and `carla-ros-bridge`) are inside the same apt repository and all the future updates will be added in this same repository. Follow the next instructions to install the packages:

```
sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-keys 1AF1527DE64CB8D9
sudo add-apt-repository "deb [arch=amd64] http://dist.carla.org/carla $(lsb_release -sc) main"

sudo apt-get update
sudo apt-get install carla-simulator
sudo apt-get install carla-ros-bridge
```

I would recommend removing all the entries pointing to CARLA apt repositories from `/etc/apt/sources.list` to avoid any conflict. Please, note that the instructions above will install the lastest version of CARLA (i.e., pre-release 0.9.10).

Feel free to re-open the issue if needed.

closed 2020-08-31 by joel-mb