carla-simulator/carla #2956: run carla-0.9.9 ./CarlaUE4.sh crash

whuxhzhang, 2020-06-16:
4.24.3-0+++UE4+Release-4.24 518 0
Disabling core dumps.
LowLevelFatalError [File:Unknown] [Line: 772] 
VulkanRHI::vkCreateBufferView(GetParent()->GetInstanceHandle(), &ViewInfo, VULKAN_CPU_ALLOCATOR, &View) failed, VkResult=-2
 at /home/zhangxinhong/UnrealEngine_4.24/Engine/Source/Runtime/VulkanRHI/Private/VulkanRHI.cpp:1423 
 with error VK_ERROR_OUT_OF_DEVICE_MEMORY
Signal 11 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554 
CommonUnixCrashHandler: Signal=11
Malloc Size=65535 LargeMemoryPoolOffset=131119 
Malloc Size=127232 LargeMemoryPoolOffset=258368 
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.

ubuntu18.04,
carla-0.9.9
unreal_engine4.24

hh0rva1h, 2020-06-17:
Try to run it with -opengl.

nuomizai, 2020-06-18:
> Try to run it with -opengl.

Hey, @hh0rva1h . I also met this problem. This didn't work for me. I using ubuntu16.04 and carla-0.9.8. I have cleaned up the display by running "DISPLAY= ./CarlaUE4.sh", as suggested by the carla tutorial. I connected to the server by vnc service by the way.

germanros1987, 2020-06-25:
@whuxhzhang are you still having this problem?

whuxhzhang, 2020-07-01:
> @whuxhzhang are you still having this problem?

the problem has been solved.

closed 2020-07-01 by whuxhzhang

tinmodeHuang, 2020-08-20:
> > @whuxhzhang are you still having this problem?
> 
> the problem has been solved.

hi @whuxhzhang 
could you share cause of the crash and your solution with me?

whuxhzhang, 2020-08-20:
> > > @whuxhzhang are you still having this problem?
> > 
> > 
> > the problem has been solved.
> 
> hi @whuxhzhang
> could you share cause of the crash and your solution with me?

I am not sure the cause of the problem, and I  executed the following order：
wget https://carla-releases.s3.eu-west-3.amazonaws.com/Linux/UE_Patch/430667-13636743-patch.txt ~/430667-13636743-patch.txt
patch --strip=4 < ~/430667-13636743-patch.txt

then 
./Setup.sh && ./GenerateProjectFiles.sh && make
These command is executed in UE4 folder.

humbert2018, 2021-01-07:
> Try to run it with -opengl.

really works!