carla-simulator/carla #2958: make Libcarla.client.rss error:- c++: internal compiler error: Killed (program cc1plus)

Akshay-sudo-ux, 2020-06-16:
Hi, I am using a machine with the following specs:-
Ubuntu - 18.04, Carla - 0.9.9 RAM 8GB Graphics - 4GB GPU.

I faced an issue with make LibCarla.client.rss:- The log messages are as below.
Ad-rss.sh: Compiling ad-rss-3.0.0.
Starting >>> spdlog  
Starting >>> ad_map_opendrive_reader
Finished <<< ad_map_opendrive_reader [15.4s]                                                              
Finished <<< spdlog [18.0s]                         
Starting >>> ad_physics
[Processing: ad_physics]                             
Finished <<< ad_physics [41.1s]                         
Starting >>> ad_map_access
Starting >>> ad_rss
[Processing: ad_map_access, ad_rss]                                                                 
[Processing: ad_map_access, ad_rss]                                                                         
Finished <<< ad_rss [1min 21s]                                                                                   
[Processing: ad_map_access]                                       
[Processing: ad_map_access]                                       
[Processing: ad_map_access]                                       
[Processing: ad_map_access]                                       
--- stderr: ad_map_access                                         
**c++: internal compiler error: Killed (program cc1plus)**
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
make[3]: *** [python/CMakeFiles/ad_map_access_python3.dir/AdMapAccessPythonWrapper.cpp.o] Error 4
make[2]: *** [python/CMakeFiles/ad_map_access_python3.dir/all] Error 2
make[2]: *** Waiting for unfinished jobs....
In file included from /home/build/carla/Build/boost-1.72.0-c8-install/include/boost/preprocessor/repetition/enum.hpp:17:0,
                 from /home/build/carla/Build/boost-1.72.0-c8-install/include/boost/preprocessor/enum.hpp:15,
                 from /home/build/carla/Build/boost-1.72.0-c8-install/include/boost/function/detail/prologue.hpp:20,
                 from /home/build/carla/Build/boost-1.72.0-c8-install/include/boost/function/function_template.hpp:13,
                 from /home/build/carla/Build/boost-1.72.0-c8-install/include/boost/function/detail/maybe_include.hpp:15,
                 from /home/build/carla/Build/boost-1.72.0-c8-install/include/boost/function/function0.hpp:11,
                 from /home/build/carla/Build/boost-1.72.0-c8-install/include/boost/python/errors.hpp:13,
                 from /home/build/carla/Build/boost-1.72.0-c8-install/include/boost/python/handle.hpp:11,
                 from /home/build/carla/Build/boost-1.72.0-c8-install/include/boost/python/args_fwd.hpp:10,
                 from /home/build/carla/Build/boost-1.72.0-c8-install/include/boost/python/args.hpp:10,
                 from /home/build/carla/Build/boost-1.72.0-c8-install/include/boost/python.hpp:11,
                 from /home/build/carla/Build/ad-rss-3.0.0/build/ad_map_access/python/AdMapAccessPythonWrapper.cpp:13:
/home/build/carla/Build/ad-rss-3.0.0/build/ad_map_access/python/AdMapAccessPythonWrapper.cpp: In function ‘void init_module_libad_map_access_python()’:
/home/build/carla/Build/ad-rss-3.0.0/build/ad_map_access/python/AdMapAccessPythonWrapper.cpp:870:1: **note: variable tracking size limit exceeded with -fvar-tracking-assignments**, retrying without
 BOOST_PYTHON_MODULE(libad_map_access_python){
 ^
make[1]: *** [all] Error 2
---
Failed   <<< ad_map_access [3min 29s, exited with code 2]

Summary: 4 packages finished [4min 28s]
  1 package failed: ad_map_access
  1 package had stderr output: ad_map_access
  1 package not processed
Util/BuildTools/Linux.mk:138: recipe for target 'ad-rss' failed
make: *** [ad-rss] Error 2

I am suspecting that this issue is related to minimum RAM requirements to integrate RSS with carla since I observed that the cc1plus process was running with two instances and each instance used more than 45% of RAM while running the make LibCarla.client.rss command.
Can you please provide a solution to this issue?

germanros1987, 2020-06-25:
@berndgassmann could you please have a look at this?

berndgassmann, 2020-06-26:
Hi,
can you try adding 
 --executor sequential
to the colcon build command within Ad-rss.sh 
https://github.com/carla-simulator/carla/blob/master/Util/BuildTools/Ad-rss.sh#L80
I believe, this might solve the issue.

PR #3024 (2020-07-14): Update to AD RSS Library 4.0.0
#### Description

- Added support for unstructured scenes and pedestrians.
- Rework rss example folder structure
- Python based debug drawing
- Handle negative speeds
- Fixes #2871 by linking libproj statically.
- Fixes #2958 by using sequential colcon build
- Allow setting log level
- Add ActorConstellationCallback
- Move setting of RssDynamics to python code

#### Where has this been tested?

  * **Platform(s):** ...
  * **Python version(s):** ...
  * **Unreal Engine version(s):** ...

#### Possible Drawbacks

---
This change is [[image]](https://reviewable.io/ [... cut]
commit 88ae9d04 (2020-07-14): Update to AD RSS Library 4.0.0

- Added support for unstructured scenes and pedestrians.
- Rework rss example folder structure
- Python based debug drawing
- Handle negative speeds
- Fixes #2871 by linking libproj statically.
- Fixes #2958 by using sequential colcon build
- Allow setting log level
- Add ActorConstellationCallback
- Move setting of RssDynamics to python code
closed 2020-07-14 by marcgpuig