carla-simulator/carla #2963: RoadRunner 2019.2.12 on Ubunty 18.04 black screen

mvdstelt, 2020-06-17:
Tho whom it may concer,

I am running Ubuntu 18.04 and have installed in this distro VerctorZero Roadrunner 2019.2.12.
After starting RoadRunner without issues and going to the Editor page, the editor page is only black.
This also happens with the RoadRunner 2019.2.11 version.
last week it was working, untilll Ubuntu crashed and I had tore-install Ubuntu and all Apps.
I am also running Unity3D, but now via the UnityHub.
Does anybody know how to solve the Black Screen issue for RoadRunner in Ubuntu?

[image]

germanros1987, 2020-06-25:
Please, talk directly with VectorZero (now Mathworks). They should be able to help you.

closed 2020-06-25 by germanros1987

mvdstelt, 2020-06-26:
Thanks.

Kind regards and thanks,
Machiel van der Stelt

https://stelt-technologies.com.au/
https://archive.org/details/@machiel_van_der_stelt