carla-simulator/carla #2964: windows10 remote access ubuntu20.4 install unrealengine issues: Error: FLinuxApplication::CreateLinuxApplication() : InitSDL() failed, cannot create application instance.

gengtsh, 2020-06-18:
I’m trying to install Carla in Ubuntu20, but i’m stuck in install unrealengine,
when i run ./UE4Editor, i got the following issues:

[2020.06.18-06.13.57:491][ 0]LogInit: Initializing SDL.
[2020.06.18-06.13.59:268][ 0]LogInit: Warning: Could not initialize SDL: XRandR support is required but not available
[2020.06.18-06.13.59:268][ 0]LogInit: - Physical RAM available (not considering process quota): 12 GB (11978 MB, 12265568 KB, 12559941632 bytes)
[2020.06.18-06.13.59:269][ 0]LogInit: - VirtualMemoryAllocator pools will grow at scale 1.4
[2020.06.18-06.13.59:269][ 0]LogInit: - MemoryRangeDecommit() will be a no-op (re-run with -vmapoolevict to change)
[2020.06.18-06.14.00:120][ 0]LogInit: Using OS detected language (en-US).
[2020.06.18-06.14.00:120][ 0]LogInit: Using OS detected locale (en-US).
[2020.06.18-06.14.00:120][ 0]LogTextLocalizationManager: No specific localization for ‘en-US’ exists, so the ‘en’ localization will be used.
[2020.06.18-06.14.00:296][ 0]LogInit: Initializing SDL.
[2020.06.18-06.14.01:478][ 0]LogInit: Warning: Could not initialize SDL: XRandR support is required but not available
[2020.06.18-06.14.01:478][ 0]LogInit: Warning: FLinuxSplashState::InitSplashResources() : InitSDL() failed, there will be no splash.
[2020.06.18-06.14.01:478][ 0]LogInit: Initializing SDL.
[2020.06.18-06.14.03:372][ 0]LogInit: Warning: Could not initialize SDL: XRandR support is required but not available
[2020.06.18-06.14.03:372][ 0]LogInit: Error: FLinuxApplication::CreateLinuxApplication() : InitSDL() failed, cannot create application instance.
[2020.06.18-06.14.03:372][ 0]LogCore: FUnixPlatformMisc::RequestExit(bForce=true, ReturnCode=1)
[2020.06.18-06.14.03:372][ 0]LogCore: FUnixPlatformMisc::RequestExit(1)

my environment:
win10 -> xmanager (xshell) -> ubuntu 20.4 server

gengtsh, 2020-06-18:
I have found the solution: 
I forget to set Two TCP ports and good internet connection. 2000 and 2001 in my xshell ssh tunneling

closed 2020-06-18 by gengtsh

ArghyaPal, 2020-08-20:
Did you solve the issues?

knight37x, 2020-11-05:
I would be interested in the solution as well. Did anyone find a solution?

ranimg, 2022-05-16:
What is the solution for this? Even Im facing this in when launching carla server or during make launch