carla-simulator/carla #2966: Traffic Manager Problems

AlexScheid, 2020-06-18:
Hello Carla-Community,

I want to use the TrafficManager on my own map in the OpenDRIVE standalone mode to simulate some traffic. During some first tests with a single vehicle some problems occurred:

1. The vehicle did not change the lane in a curve with 2 existing lanes, both in the driving direction of the vehicle. The lanes are existing and the Client can get access to both of them. I checked that:
```
loc = self.vehicle.get_location()
waypoint = self.map.get_waypoint(loc, True)
left_lane = waypoint.get_left_lane()
right_lane = waypoint.get_right_lane()
```
Either right_lane or left_lane contains a waypoint, depending on the current lane of the vehicle. 

2. I want to change the velocity of the vehicle at specific positions. These informations are not provided in the .xodr-file, where the speed limit is defined (if I understand it right), therefore I want to change the percentage speed difference of the vehicle with the command:
`self.traffic_manager.vehicle_percentage_speed_difference(self.vehicle, self.speed_percentage)`
But this did not affect anything. Is there another way to perfom such a behaviour?

jackbart94, 2020-06-19:
Hi,

Thanks for using the Traffic Manager, I hope I can shed some light on your doubts.

1. Traffic Manager has three different parameters for lane changes. The first one, which is active by default, is the automatic one, which basically activates lane change in different situations (for example, a car slower is in front so we change the lane to pass it). The one you might be looking for is the forced lane change (check among the parameters!) where you can decide to have the car move to the lane on the right or left at runtime. In addition to this, there's a "keep right" rule parameter, which is by default set to 0, where you can specify how much a car is likely to stay in the most-right lane. Finally, you can change the code yourself for the automatic lane change if you want a specific policy!

2. Setting the velocity in the Traffic Manager works in this way: you always have a speed limit affecting the vehicle, and you set the percentage of speed difference that you want your vehicle to go with respect to that limit. There's a vehicle-specific method (the one you are using) and also a global method, including all vehicles under that specific Traffic Manager, which is called ``global_percentage_speed_difference``. 

Let me know if this helps you!

AlexScheid, 2020-06-19:
Hi,

thank you for the quick response!

1. For the point of time I want to force a lane change of a specif vehicle at a given location, therefore I turned off the automatic lane changes and force a lane change with:
`self.traffic_manager.force_lane_change(self.vehicle, direction)`
This command is working as expected, when my vehicle is on a straight road. But when it is driving in a curve it does not react on this command. I checked the map by using `waypoint.lane_change` and `waypoint.get_left_lane()` / `waypoint.get_right_lane()` and with both variants I get access to a neighbor lane in the expected direction. It is a quite long curve, which should give enough waypoints for the route planning. Do you have any clue what's going wrong in this case?

2. Because my traffic manager should handle a small swarm of vehicles with different velocities I would set the velocity for each vehicle with:
`self.traffic_manager.vehicle_percentage_speed_difference(self.vehicle, speed_percentage)`
At a specific location I want to change the velocity of a vehicle by applying the same command with the same vehicle and a different value for the variable `speed_percentage`. I tested this on a straight road, which is long enough to see some reactions, but they did to appear. Is it possible to change the percentage speed difference of a vehicle from time to time by using the traffic manager? I also looked at the last throttle command given by the traffic manager, which stays constant after a change of the speed percentage value. In this context I assume the throttle value is capped by 0.6999.., can it be changed by the PythonAPI?

germanros1987, 2020-06-25:
@jackbart94 could you follow up?

AlexScheid, 2020-06-26:
Concerning point 1: I take a closer look at the used map and I think the problem can be caused by the definition of the curve. It is assigned to a junction, what surprised me a bit, because it is a very long curve. I think lane changes are not possible at a junction, if I remember that right, therefore this part of my problem can be seen as solved. Sorry for that.

For the second question/problem I have not been able to find a solution.

jackbart94, 2020-06-26:
Yeah, you're absolutely right about point 1.

For point 2, throttle is indeed capped at 0.7, but you'll have to change it in the Traffic Manager API. You just have to modify the constant MAX_THROTTLE in Constants.h and do ``make LibCarla`` to apply the changes. Having said so, it is possible to change the values of maximum velocity for vehicles at different times in the execution, via the PythonAPI as you're doing, but I'll need some more information (and possibly the code to test it on my computer) because it could be possible that, for example, you aren't setting the conditions appropriately.

AlexScheid, 2020-06-29:
Thank you for the hint regarding the MAX_THROTTLE.
I also thought that it is possible to change the velocity in this way, but it does not affect anything. A short version of my problem with only one vehicle is given by the code below. It is on a map, which shows a highway situation.
`import glob
import os
import sys

try:
    sys.path.append(glob.glob('../carla/dist/carla-*%d.%d-%s.egg' % (
        sys.version_info.major,
        sys.version_info.minor,
        'win-amd64' if os.name == 'nt' else 'linux-x86_64'))[0])
except IndexError:
    print('failed to import carla')
    print(sys.version_info.major)
    print(sys.version_info.minor)
    print('win-amd64' if os.name == 'nt' else 'linux-x86_64')

import carla
import numpy as np

def get_map():
    """ read in an openDrive file"""
    data = None

    return data

def get_valid_spawn_point(world):
    """
    used for my map to get valid spawn points
    """
    all_spawn_points = world.get_map().get_spawn_points()
    spawn_points = []
    for spawn_point in all_spawn_points:
        if 250. < spawn_point.location.y < 350.:
            if 250 < spawn_point.location.x < 350.:
                spawn_points.append(spawn_point)

            elif -250 > spawn_point.location.x > -350.:
                spawn_points.append(spawn_point)
    
    return spawn_points

class SingleVehicle(object):
    """
    test class for a single vehicle
    """
    def __init__(self, client, spawn_point, default_velocity):
        """
        set default parameters and activates the tm initialization
        """
        self.client = client
        self.world = self.client.get_world()
        self.map = self.world.get_map()
        self.spawn_point = spawn_point
        self.vehicle = None

        self.traffic_manager = None
        self._initialize_tm()

        self.target_velocity = default_velocity

    def _initialize_tm(self):
        """
        initialization of the tm
        """
        self.traffic_manager = self.client.get_trafficmanager(8000)
        self.traffic_manager.set_global_distance_to_leading_vehicle(2.0)

    def _target_reached(self):
        """
[... 107 lines cut ...]
            world.tick()
            frame += 1

            

    finally:
        if vehicle:
            del vehicle
        settings = world.get_settings()
        settings.synchronous_mode = False
        settings.fixed_delta_seconds = None
        world.apply_settings(settings)

# ==============================================================================
# -- start point ---------------------------------------------------------------
# ==============================================================================

if __name__ == "__main__":
    try:
        main()
    except KeyboardInterrupt:
        print("shut down forced")

    finally:
        print("bye")`

AlexScheid, 2020-07-02:
Thank you for the hint regarding the MAX_THROTTLE.
I also thought that it is possible to change the velocity in this way, but it does not affect anything. A short version of my problem with only one vehicle is given by the code below. It is on a map, which shows a highway situation. (Maybe relevant for stable driving at this velocity)
Sorry for the bad formatting of the last post, here is the code in a txt-file.
[simple_scenario.txt](https://github.com/carla-simulator/carla/files/4862403/simple_scenario.txt)

jackbart94, 2020-07-08:
Sorry if it took a while to answer. I think there might be a bug for this particular parameter ``vehicle_percentage_speed_difference`` as it seems to be applying successfully only once. We will push a fix very soon and I'll let you know.

I've tested your script and then tried on another one to replicate the issue, and it kept happening. I'll make sure to mention this issue once the PR is open.

AlexScheid, 2020-07-08:
thank you very much for all the help!

closed 2020-09-13 by stale[bot]

dhiegomaga, 2020-10-22:
> 1. In addition to this, there's a "keep right" rule parameter, which is by default set to 0, where you can specify how much a car is likely to stay in the most-right lane.

What parameter is this? I can't find it in the documentation.

jackbart94, 2020-10-22:
> > 1. In addition to this, there's a "keep right" rule parameter, which is by default set to 0, where you can specify how much a car is likely to stay in the most-right lane.
> 
> What parameter is this? I can't find it in the documentation.

Thanks for pointing it out, it's actually not in the documentation which we must update accordingly.

Anyway, the parameter is ``set_percentage_keep_right_rule``:

```
set_percentage_keep_right_rule(self, actor, perc)

During the localization stage, this method sets the percent chance that a vehicle changes lane to the right without any traffic or path implication.

    Parameters:
        actor (carla.Actor) – The actor that is going to move to the right lane.
        perc (float) – Between 0 and 100, chance that the actor might move to the right lane.
```

jmoss7, 2024-06-08:
Hi @jackbart94 , Do you know if multiple calls to force_lane_change() will result in the same number of lane changes? I'm trying to do something where my agents perform X number of lane changes. I have a behavior tree for each vehicle, where X number of ForceLaneChangeBehavior nodes are added upon creation of their trees. I'm trying to have the nodes only return SUCCESS after the vehicle is fully in the new lane, so additional calls to force_lane_change don't happen prematurely. 

Right now, I can't seem to get the vehicles to do more than one lane change in quick succession.

Any help is greatly appreciated!
Thanks,
John

PR #3262 (2020-09-17): Fixes add entry issue for applying parameters more than once
#### Description

Changes in Traffic Manager to fix an issue related to applying specific parameters more than once in the same execution.

Fixes #3224
Fixes #2966

#### Where has this been tested?

  * **Platform(s):** Ubuntu 18
  * **Python version(s):** 3.6
  * **Unreal Engine version(s):** 4.24

#### Possible Drawbacks

None

---
This change is [[image]](https://reviewable.io/reviews/carla-simulator/carla/3262)