carla-simulator/carla #2968: How to use multiprocessing to achieve multi-sensor simultaneous collect data without causing resource competition

JarvanI, 2020-06-19:
Hi everyone，I‘m trying to use Carla to test our autonomous driving system. I need to generate 5 lidars and 8 cameras and make them work together, but I found that client is single process. I tried to spawn 5 lidars and listened simultaneously,  the result was bad. Some data was lost and some was disorder. It looks like sensors are competing for resources. So I tried to use multiprocessing module to solve this problem. I planned to create a process for each sensor but failed. Here‘s my code 
`p = Process(target=sensors_actor.listen, args= lambda image: self._parse_image(image))
 p.start()`
Here is error report
[image]
I want to know how to create a process for listen function? It has tormented me for days. Thanks!

germanros1987, 2020-06-25:
Hi @JarvanI 

Thanks for reaching out. The setup that you are trying to create is far from trivial. It is a very demanding sensor stack, which will require the right hardware to run in real-time. It will also require an upgrade in the CARLA internal architecture to leverage multiple GPUs. This upgrade will happen at some point during 0.9.11.

Now, if you don't care about real-time, the setup you are describing is totally doable. You don't need to go for multithreading or anything similar. Just follow the standard procedure to create sensors and fetch data. Of course, this will be slow, but using multiple processes in client will not make it faster, since the bottleneck is not there.

JarvanI, 2020-06-26:
Thank you @germanros1987 ! I gonna try it and I'm really looking forward to 0.9.11. I have found that I made a stupid mistake that computation is excuted in server

closed 2020-06-26 by JarvanI

svanimisetti, 2020-12-11:
@germanros1987 - after some experimentation I came to the same conclusion. Using multiprocessing on the client side with one worker per sensor did not improve performance. There seem to be three issues: (1) the carla.SensorData objects are not picklable and therefore raw data has to be sent to each worker process (2) all sensor data passed through a single TCP/IP connection - so this may be the bottleneck (3) CARLA server and RPC calls may be running on a single thread(??).

So my question is - is there a way to have multiprocessing implemented on the server side - and have the ability to send image sensor data over multiple TCP/IP connection? This will perhaps improve performance - and give the ability to use multiprocessing with queues / shared NP-arrays on the client side?