carla-simulator/carla #2970: error: XDG_RUNTIME_DIR not set in the environment. in WSL Ubuntu 18.04

jportella91, 2020-06-19:
When I run ./CarlaUE4.sh in the WSL, I obtain the following error:

```
4.22.3-0+++UE4+Release-4.22 517 0
Disabling core dumps.
error: XDG_RUNTIME_DIR not set in the environment.
CarlaUE4-Linux-Shipping: ../nptl/pthread_mutex_lock.c:434: __pthread_mutex_lock_full: Assertion `robust || (oldval & FUTEX_OWNER_DIED) == 0' failed.
Signal 6 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554
CommonUnixCrashHandler: Signal=6
Malloc Size=65535 LargeMemoryPoolOffset=131119
Malloc Size=99968 LargeMemoryPoolOffset=231104
Engine crash handling finished; re-raising signal 6 for the default handler. Good bye.
Aborted (core dumped)
```

I am new to WSL and Ubuntu. So I would appreciate a lot if you can explain me why is this error happening and how to fix it please. It is Carla 09.7 version

nuomizai, 2020-06-22:
Hey, @jportella91 . I met the same problem when I ran CarlaUE4.sh in a server without ubuntu destop. The system is Ubuntu16.04. Unfortunately, I didn't fix that problem. But I ran the CarlaUE4.sh successfully in a destop computer. So I guess CarlaUE4.sh can not run in a headless system without a destop. So I installed carla in the docker and it works well with the server. You can look that in https://carla.readthedocs.io/en/latest/build_docker/.  Or you can try to install ubuntu-desktop. But I found it not easy to install a ubuntu-desktop and I failed. If you had to try the second instruction, then good luck :)

germanros1987, 2020-06-25:
@nuomizai @jportella91 CARLA can run headless in some configurations, but it seems that WSL is not one of them. 
We have never testes a WSL setup...so chances are that this will not work. Could you try with the docker option, and check if that works?

closed 2020-08-29 by germanros1987

blackliner, 2020-09-11:
Hi, with WSL2 being able to run nvidia-docker2 with GPU support, I tried to get carla running. Unfortunately, I get an error message and directly thrown out of the docker again:
```
$ docker run --gpus=all -it carlasim/carla
4.24.3-0+++UE4+Release-4.24 518 0
Disabling core dumps.
sh: 1: xdg-user-dir: not found
$
```

perseus784, 2020-10-06:
I too am facing the same issue as of @blackliner. Please let us know if you have found a solution to run it on wsl.

AN-Ahmed, 2020-10-23:
Have anyone reached a solution regarding that?

blackliner, 2021-03-03:
For now I start the `CarlaUE4.exe` in Windows and the `carla-ros-bridge` in WSL2, works like a charm

Jags27GitHub, 2022-06-29:
> For now I start the `CarlaUE4.exe` in Windows and the `carla-ros-bridge` in WSL2, works like a charm

I tried the same i.e launching CARLA on windows and trying to start bridge on WSL2. But i get the following error. Any idea ? What is wrong ? i appreciate any inputs.

[ERROR] [1656482911.167891, 0.000000]: Error: time-out of 2000ms while waiting for the simulator, make sure the simulator is ready and connected to localhost:2000