carla-simulator/carla #2971: Nothing happens when executing the example python scripts

made3, 2020-06-20:
Hello,

I have gone through both, the Quick start installation, as well as the Windows build. When I come to the point where I am supposed to spawn the npcs with the spawn_npc.py script it won't do anything.

I use the command
`py spawn_npc.py`
because I can execute a normal helloworld.py file and will get the output of it.

Now after I enter the command into the cmd there is no output, nothing at all. 

When I entered the make PythonAPI command I had errors [like this](https://github.com/carla-simulator/carla/issues/2339#issuecomment-571965712). I tried to fix these but nothing worked, but for the Quick Installtion guide I don't even need to do that, right?

Thank you already in advance for your help

closed 2020-06-20 by made3

xmgpev, 2020-08-29:
Hi,

I have the same problem but with a similar configuration: Python 3.7, Carla 0.9.9, Win10. 
Did you solved the issue?

Thanks

ziyanan, 2020-09-09:
How did you guys solve this? I'm currently stuck on the same problem for as well.

BAROBA-3d, 2020-12-10:
Hi all,
I think you need to start the pythonscript from <carladir>\pythonapi\examples\ or the dir where you put spawn_npc.py.
Then I can start the script but not Carla..Maybe I need to copy the scripts to somewhere else?
Report back if you get it to work :)