carla-simulator/carla #2972: How to run Scenario Runner

Mechatrnk, 2020-06-20:
Hello, I [downloaded the Scenario Runner for CARLA 0.9.9](https://github.com/carla-simulator/scenario_runner/releases/tag/v0.9.9) and I am trying to run CARLA Scenario Runner, following [these instructions](https://github.com/carla-simulator/scenario_runner/blob/master/Docs/getting_started.md). I am stuck in part Running the follow vehicle example when I try to run `export CARLA_ROOT=/path/to/your/carla/installation` I get an error that export is `not supported external or internal command.`

I am running it via cmd from the folder where the `scenario_runner.py` is saved I am using Windows and Python 3.7 (I know that Python 3.5 is requested, but in my humble opinion this is not the reason for the error). I also tried to run it using `x64 Native Tools Command Prompt for VS 2019` but I got the same error.

Do you have any ideas on how can I fix that?

Thank you very much

germanros1987, 2020-06-25:
@glopezdiest could you please help this user?

glopezdiest, 2020-06-26:
Hey @Mechatrnk The GettingStarted is only explained for Ubuntu, not for Windows, that's why you are seeing these issues. I actually just found about this yesterday, and we'll be changing the Docs shortly. 

**For Windows, the environment variables** (CARLA_ROOT and PYTHONAPI) **have to be set up different:**

1) Go to "edit the environment variables of this account" (can be access by writing *env* on the search panel).
2) Create two new variables (either for the user of the system, depending on whether or not you use multiple accounts / have admin priviliges):
   · CARLA_ROOT, which will be mapped to your CARLA folder
   · PYTHON_PATH, mapped to ${CARLA_ROOT}/PythonAPI/carla/dist/carla-<VERSION>.egg; ${CARLA_ROOT}/PythonAPI/carla/agents; ${CARLA_ROOT}/PythonAPI/carla

For the PYTHON_PATH, **use ";" instead of ":"** to concatenate different paths. I'm not sure if the * ${CARLA_ROOT}* works for Windows, so it might be a good idea to just specify the paths directly. I'll update this when I figure it out.

Hope this helps.

A couple of notes. I'm gonna change the title and transfer the issue to the ScenarioRunner repo, so that other people can see it there. I'll close this when we update the Docs.