carla-simulator/carla #2974: IMU and GNSS data fusion Using EKF

Gilaine, 2020-06-20:
I am trying to fuse the IMU data with GNSS data Using Extended kalman filter, but when I decrease the IMU variance, the estimated data diverge which mean that the IMU data is not accurate, is this correct or maybe I'm doing something wrong in my EKF filter?

Also, I have to decrease the GNSS variance to 0.001 and increase the IMU variance to 0.1 and higher so that the estimated data converge and have a good accuracy.

This is example of what I have " Camera + IMU + GNSS fusion ":

var_imu = 0.01
var_gnss = 0.0001
var_cam = 0.1 

[image]

GabiShahmayster, 2020-06-20:
> I am trying to fuse the IMU data with GNSS data Using Extended kalman filter, but when I decrease the IMU variance, the estimated data diverge which mean that the IMU data is not accurate, is this [...]

Hi Gilaine,
One possible explaination would be related to the EKF covariances. Typically, providing a KF-based solution with more accurate measurements (smaller measurement covariance), or more accurate process (less process noise), results in a smaller filter covariance, this is when the filter "indifferent" towards new measurements and "trusts" it's predictions more than the measurements... in this case, you would see the filter innovations multiplied by a very low Kalman gain, i.e the measurement-update stage would have very little effect, compared to the time-update stage

Gilaine, 2020-06-25:
I try to do this right now but still i'm not sure.

germanros1987, 2020-06-25:
Hi @Gilaine, 

This seems like a stackoverflow question. It doesn't look like this should take place in our Github issues. You can also ask in the CARLA forum: forum.carla.org, but this definitely does not belong in Github. We are trying to keep this section for feature requests and bugs.

closed 2020-06-25 by germanros1987