carla-simulator/carla #2975: nothing happened when run carla in docker

nuomizai, 2020-06-21:
I successfully pulled the carlasim/carla:0.9.7. After I ran the command

docker run -d carlasim/carla:0.9.7

no container id returned

[image]

And I ran the command "docker ps -a". But there was no containers!
[image]

umateusz, 2020-06-21:
The missing container id is strange. I'm not sure where it might come from. Do you have nvidia gpu support installed in the docker?

You should start Carla according to the instructions:
https://carla.readthedocs.io/en/latest/build_docker/

E.g.:
`docker run -p 2000-2002:2000-2002 --runtime=nvidia --gpus all carlasim/carla:0.8.4`

You must install nvidia support for the docker. I think the documentation is a bit outdated. You can try to install: 
`https://github.com/NVIDIA/nvidia-docker`
and run Carla using:
`docker run -p 2000-2002:2000-2002 --gpus=all carlasim/carla:0.9.7`

If you want to run Carla in a docker without a GPU, see:
https://github.com/carla-simulator/carla/issues/2899

nuomizai, 2020-06-22:
Thank you, @umateusz . Yes, I have installed nvidia gpu sopport correctly. I have tested nvidia support by the command:

`sudo docker run --gpus all nvidia/cuda:10.0-base nvidia-smi`

And the result was:

Unable to find image 'nvidia/cuda:10.0-base' locally
10.0-base: Pulling from nvidia/cuda
7ddbc47eeb70: Pull complete
c1bbdc448b72: Pull complete
8c3b70e39044: Pull complete
45d437916d57: Pull complete
d8f1569ddae6: Pull complete
de5a2c57c41d: Pull complete
ea6f04a00543: Pull complete
Digest: sha256:e6e1001f286d084f8a3aea991afbcfe92cd389ad1f4883491d43631f152f175e
Status: Downloaded newer image for nvidia/cuda:10.0-base
Mon Jun 22 07:51:38 2020
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 440.33.01    Driver Version: 440.33.01    CUDA Version: 10.2     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|===============================+======================+======================|
|   0  Tesla P40           On   | 00000000:84:00.0 Off |                  Off |
| N/A   20C    P8     9W / 250W |      0MiB / 24451MiB |      0%      Default |
+-------------------------------+----------------------+----------------------+

+-----------------------------------------------------------------------------+
| Processes:                                                       GPU Memory |
|  GPU       PID   Type   Process name                             Usage      |
|=============================================================================|
|  No running processes found                                                 |
+-----------------------------------------------------------------------------+

Today I tried to run carla image again
`docker run -p 2000-2002:2000-2002 --runtime=nvidia --gpus all carlasim/carla:0.9.7`
but it still had no response. So I stopped the process by CTRL+C
But after I ran the command

`docker ps -a`

The result was:

luban@k8s-deploy-sx4fqm-1592623066130-6dfc566ddb-n984v:~$ docker ps -a
CONTAINER ID        IMAGE                   COMMAND                  CREATED             STATUS                     PORTS                              NAMES
0bdb91ba3c59        nvidia/cuda:10.0-base   "nvidia-smi"             10 minutes ago      Exited (0) 8 minutes ago                                      ecstatic_northcutt
2efdbea8d30b        carlasim/carla:0.9.7    "/bin/sh -c '/bin/ba…"   3 hours ago         Created                    0.0.0.0:2000-2002->2000-2002/tcp   jovial_gates

**The container's status was Created.**  And I tried to restart the container and ran the command

`docker restart jovial_gates`

the result was

luban@k8s-deploy-sx4fqm-1592623066130-6dfc566ddb-n984v:~$ docker restart jovial_gates
Error response from daemon: Cannot restart container jovial_gates: OCI runtime create failed: unable to retrieve OCI runtime error (open /run/containerd/io.containerd.runtime.v1.linux/moby/2efdbea8d30b1b6439ab1a4b87d3d1502c39b0a0c1cc6352a62585087ce1e0ac/log.json: no such file or directory): fork/exec /usr/bin/nvidia-container-runtime: no such file or directory: unknown

And I can't figure out the reason.

nuomizai, 2020-06-22:
> The missing container id is strange. I'm not sure where it might come from. Do you have nvidia gpu support installed in the docker?
>
> You should start Carla according to the instructions: [...]

Hey, @umateusz . I just ran carla image for another. And this time, after I ran the command:
`docker run -p 2000-2002:2000-2002 --gpus=all carlasim/carla:0.9.7 `

The result was:
sh: 1: xdg-user-dir: not found

And after I ran the command
`docker ps`

The result was:
CONTAINER ID        IMAGE                  COMMAND                  CREATED             STATUS              PORTS                              NAMES
d76d966907b3        carlasim/carla:0.9.7   "/bin/sh -c '/bin/ba…"   25 minutes ago      Up 8 minutes        0.0.0.0:2000-2002->2000-2002/tcp   inspiring_booth

After I ran the command
`docker ps -a`

The result was:
CONTAINER ID        IMAGE                   COMMAND                  CREATED             STATUS                      PORTS                              NAMES
d76d966907b3        carlasim/carla:0.9.7    "/bin/sh -c '/bin/ba…"   22 minutes ago      Up 4 minutes                0.0.0.0:2000-2002->2000-2002/tcp   inspiring_booth
0bdb91ba3c59        nvidia/cuda:10.0-base   "nvidia-smi"             29 minutes ago      Exited (0) 28 minutes ago                                      ecstatic_northcutt
2efdbea8d30b        carlasim/carla:0.9.7    "/bin/sh -c '/bin/ba…"   4 hours ago         Created                                                        jovial_gates

Did these results mean that the carla image worked successfully? I also wanted to figure out how to fix the error "sh: 1: xdg-user-dir: not found". Do you have any idea? Thank you!

umateusz, 2020-06-25:
Do not worry about `sh: 1: xdg-user-dir: not found`. This is the message that appears when the container starts (it would probably be better to suppress this error).

Everything seems to be working properly. You can now try to connect to the simulator.

https://carla.readthedocs.io/en/stable/connecting_the_client/#running-the-client-example

You can try to connect using the `manual_control.py` script from PythonAPI. To get PythonAPI, do the following:

`docker cp <carla-container-name-or-id>:/home/carla/PythonAPI ./`

Then you can run the script:
`cd PythonAPI/examples`
`python manual_control.py`

germanros1987, 2020-06-25:
That is correct. Don't worry about the xdg error. Just connect using the manual_control.py. It should work fine.
Remember to share the network between both processes: --net=host

closed 2020-07-30 by germanros1987

Michal-Kolomanski, 2021-02-21:
> That is correct. Don't worry about the xdg error. Just connect using the manual_control.py. It should work fine.
> Remember to share the network between both processes: --net=host

For me, a connection using  **manual_control.py** doesn't work as I wish. 
I described the problem in issue #3873.
I will be very grateful for any help or tip.

ahmedyossef788, 2023-12-13:
can i run carla on the ietson nano ??