carla-simulator/carla #2976: Issues with OpenDrive files

avishek-mondal, 2020-06-21:
tldr : I have a xodr file that I generated with Sumo Netconvert, and a fbx file that I generated using Blender. When trying to use it on Carla 0.9.9, the wheels of the vehicle "sink" into the road, and I am unable to drive the vehicle or use autopilot for the vehicle spawned (using the script PythonAPI/examples/manual_control.py as the client). However, when I remove the \<link\> elements and its contents in the xodr file, I am able to drive the vehicle using W-A-S-D, but not on autopilot as the resulting Carla Map returns no waypoints when get_topology() is called. I want to be able to use autopilot - how do I resolve this? 

Hi, 

I am trying to load a custom map that I created. I'll describe the process I'm following - 

1) Generate the xodr file using sumo netconvert, and the fbx using Blender. I put both these files in  the Import directory, for example, called Import/custom_maps/keble (because I am experimenting with several maps, I'll be creating new fbx and xodr files and putting them in separate directories, like for example in Import/custom_maps/keble_1 and Import/custom_maps/keble_2 etc.) of the directory that contains the Carla source, following what is shown here https://carla.readthedocs.io/en/latest/tuto_A_add_map/ 

2) I run 
``` 
make import
```
which generates a map_package.json file in the Import directory 

3) I then run 
```
make package ARGS="--packages=map_package"
``` 
which then creates a map_package_0.9.9-dirty.tar.gz in the Dist directory

4) I then move this tar file over to the Import directory of in the directory that contains the Carla binaries from where I'm going to start the Carla server, and run ./ImportAssets.sh in the Carla binaries directory to generate all the maps

5) I then start the server using 
```
SDL_VIDEODRIVER=offscreen SDL_HINT_CUDA_DEVICE=0 ./CarlaUE4.sh -ResX=800 -ResY=600 -nosound -windowed -opengl
``` 
6) I go back to the source directory, and set the map I want to run from PythonAPI/util/config.py, and then I run PythonAPI/examples/manual_control.py, and this is what I observe - 
[image]
(another screenshot with better lighting)
[image]
I cannot use the W-A-S-D keys or drive on autopilot. 

**My issue is essentially the following - I need to be able to use autopilot on my original map (i.e. the fbx and original xodr file) and get rid of the "sunken wheels" problem. How do I do that?**

While trying to solve this, I noticed something strange. I did the following
1) Save the point that the vehicle spawned, and save a copy of manual_control.py where the vehicle always spawns at this point instead of randomly - 
```
carla.Transform(carla.Location(x=64.859673, y=95.698311, z=4.000000), carla.Rotation(pitch=0.000000, yaw=-22.828720, roll=0.000000))
```
2) Alter the xodr file the following way - I remove all \<link\> elements and any of the contents, so for example, anything like 
```
<link>
            <predecessor elementType="junction" elementId="1"/>
            <successor elementType="junction" elementId="2"/>
</link>
```
is removed (trivially, this can be done on VSCode's 'Find and replace'). I came to this "workaround" after many hours of experimenting, and I understand that this removes connectivity information about the resulting road network. 

3) However, when I follow the six steps that I described previously using this modified xodr file in the place of the original xodr file (the fbx file stays the same), there are no more "sunken wheels" for the vehicle that is spawned, and I can manually control the car using the W-A-S-D keys, but cannot use autopilot. When I query this new map whose xodr has all the \<link\> elements removed, methods such as get_topology etc all return empty lists. So while this technically does solve my problem of not being able to drive around, it is not an ideal solution because the car is not automatically driving. 

The other thing I observed is that when I try and visualise the original xodr file (the one that causes the "sunken wheels") in the opendrive standalone mode (using PythonAPI/util/config.py -x ) I am able to drive the car. 

Why is this behaviour being observed? How does removing the \<link\> elements allow driving to happen, and why does leaving it in somehow cause the wheels to "sink"? Any help will be appreciated. 

The original xodr file from Netconvert (in .txt) - 
[keble_13062020_normal.txt](https://github.com/carla-simulator/carla/files/4809627/keble_13062020_normal.txt)

The xodr file with link elements removed (also in .txt)
[keble_13062020_links_removed.txt](https://github.com/carla-simulator/carla/files/4809630/keble_13062020_links_removed.txt)

germanros1987, 2020-06-25:
@marcgpuig this sounds like your cup of tea.

salvatoresan, 2020-07-02:
Hi @avishek-mondal ,
I am really interested in map generation via blender, could you please tell me how you managed to generate a map?
I've tried something similar by using a plugin on UE4 but i' m not quite satisfied with the result.

avishek-mondal, 2020-07-02:
Hi @salvatoresan, this is still a work in progress! I'm aiming to share everything soon, as soon as I work out these last few issues - @marcgpuig have you had a chance to look at this issue yet? Thanks!

salvatoresan, 2020-07-02:
@avishek-mondal thanks for your reply

marcgpuig, 2020-07-03:
@avishek-mondal not yet! Although It's in my list of TODO things.

Lu7913, 2020-07-16:
Hi @avishek-mondal is there any progress? Could you share at least roughly abstract of what you did. Before i dive into Blender, I would like to know if this is even possible. I'm interested in creating .fbx files from osm. 
Thank you

avishek-mondal, 2020-07-16:
Unfortunately, no progress on this yet - the issue with the XODR is still not solved and is holding things back. @marcgpuig is looking into this!

Regarding Blender, it has a add-on that does fbx exporting, as well as an add-on for osm to Blender object import. Both are accessible through the python console.

sp062, 2020-08-14:
How to Export .xodr from SUMO netconvert. Have imported OSM file but how to get xodr file for it?

closed 2020-11-08 by stale[bot]

Hanss8, 2022-03-14:
@marcgpuig Hello, did you solve this problem please?