carla-simulator/carla #2977: BasicAgents do right/left on straights when spawning many of them

flobau, 2020-06-21:
Basically my goal is to extend the BasicAgent and override the run_step method to implement my own logic but i want to use the navigation system in BasicAgent. 

So i have the problem that my extended Agent starts steering right/left on straights, even when i spawn 4 of them.

So i tried it with BasicAgent aswell.

`for i in range(30):
            if not map.get_spawn_points():
                print('There are no spawn points available in your map/town.')

            blueprint = random.choice(world.get_blueprint_library().filter('vehicle.tesla.model3'))
            spawn_point = random.choice(spawn_points)
            # spawn_point = spawn_points[i]
            vehicle = world.try_spawn_actor(blueprint, spawn_point)

            if vehicle is not None:
                print('spawned ' + get_actor_display_name(vehicle))

                agent = BasicAgent(vehicle)
                agent_list.append(agent)
                dest_point = random.choice(map.get_spawn_points())
                _ = agent.set_destination((dest_point.location.x, dest_point.location.y, dest_point.location.z))`

When i spawn about 30 BasicAgents, it happens to them aswell. I have no clue why this happens. 
That's the loop i am using:

`while True:
            for agent in agent_list:
                control = agent.run_step(agent_list)
                agent._vehicle.apply_control(control)`

jackbart94, 2020-06-22:
Hi,

The PID is tuned for a specific threshold of FPS values, spawning too many Basic Agents will lead to an FPS drop and thus the PID values have to be changed accordingly.

germanros1987, 2020-06-25:
@flobau  We need much more information about this issue if you want us to help.

We need the version of your CARLA distribution. Are you working on synchronous mode or asynchronous mode? Are you using fixed-time or variable time?

closed 2020-09-05 by stale[bot]