carla-simulator/carla #2979: VehicleAI(Behavior tree)

xiaoyang666, 2020-06-22:
Hi,
   I always wanted to reproduce the automatic driving of the vehicle in my own scene.As far as I understand, the autonomous driving of UE4 can be constructed with a behavior tree.Does the automatic driving in Carla follow a fixed route set by people? I changed the position of Waypoints in the map, and the self-driving vehicles also changed, so I have the above questions. Or is vehicle autopilot related to UE4's behavior tree?
Thanks.

germanros1987, 2020-06-25:
@xiaoyang666 are you referring to the Autopilot in CARLA? In other words, the Traffic Manager? I don't fully understand your question.

germanros1987, 2020-07-30:
@xiaoyang666 is this still a problem?

xiaoyang666, 2020-08-01:
Sorry to reply to you so late, I already understand that the carla simulator does not implement autonomous driving through behavior trees, thank you for your reply

closed 2020-08-01 by xiaoyang666