carla-simulator/carla #2983: IndexError: bytes to write exceed buffer size

shizhuocaoa, 2020-06-23:
I installed Carla by Quick Installation 0.9.9.3. And install carla-ros-bridge as developers. When I try to test ros-bridge through roslaunch carla_ros_bridge carla_ros_bridge_with_example_ego_vehicle.launch. Some errors occur as follows

[image]
How can I fix it?  Thank you! Thank you!

germanros1987, 2020-06-25:
@joel-mb could you please ask for further information and take a look at this issue?

joel-mb, 2020-06-30:
Hi @shizhuocaoa ,
Can you provide more information about your setup? Is the manual control (without using ROS) working? Try running the `manual_control.py` script provided inside `/opt/carla-simulator/PythonAPI/examples`.

Please, take a look to the solution of this similar issue also (https://github.com/carla-simulator/carla/issues/991).

YanggangZhang, 2020-07-17:
I seemed have met the same problem,u can try change to install new pygame version using "pip install " rather than "apt".

yasser-h-khalil, 2020-07-23:
Installing pygame using "install" solved my issue too! I used the following command: `python2 -m pip install -U pygame --user`
Then running manual_control.py using python2.7 rather than python3.6.
With python3.6, I am getting ImportError even when running spawn_npc.py. However, with python2.7 it works fine.

closed 2020-10-04 by stale[bot]

xiangdongli, 2021-02-19:
@yasser-h-khalil  the command is useful :  python2 -m pip install -U pygame --user 
Thank you!