carla-simulator/carla #2984: Vulkan issue

yumianhuli2, 2020-06-23:
Hell0! I followed step
wget https://carla-releases.s3.eu-west-3.amazonaws.com/Linux/UE_Patch/430667-13636743-patch.txt ~/430667-13636743-patch.txt
patch --strip=4 < ~/430667-13636743-patch.txt

and

./Setup.sh && ./GenerateProjectFiles.sh && make

when I used cd ~/UnrealEngine_4.24/Engine/Binaries/Linux && ./UE4Editor

I got below:

[image]

maxc303, 2020-06-23:
Hi, If you are working on a device with Nvidia graphics card, you may try to install or upgrade your nvidia driver.

germanros1987, 2020-06-25:
I am closing this. Please, re-open it if your issue persists.

closed 2020-06-25 by germanros1987