carla-simulator/carla #2987: Road Runner textures not importing

avishek-mondal, 2020-06-23:
Hi, I'm following the instructions in the following link - https://carla.readthedocs.io/en/latest/tuto_A_add_props/ to try and import a scene I built in RoadRunner, and I'm not able to observe the textures I have on RoadRunner on Carla, despite using the no-carla-materials flag. Here are the steps I'm following 

1) Using RoadRunner's export to Carla option, I'm exporting everything - the xodr file, the fbx file and all the png and png_rr files containing texture and materials information, and also the xml file, and putting it all into a package directory in the Import directory. My Import directory looks something like this - 

Import
└── custom_maps
    └── rr_random_props
        ├── Asphalt1_Diff.png
        ├── Asphalt1_Norm.png
        ├── Asphalt1_Spec.png
        ├── Brick1_Diff.png_rrx
        ├── Brick1_Norm.png_rrx
        ├── Brick1_Spec.png_rrx
        ├── Concrete1_Diff.png
        ├── Concrete1_Norm.png
        ├── Concrete1_Spec.png
        ├── Concrete3_Diff.png_rrx
        ├── Concrete3_Norm.png_rrx
        ├── Concrete3_Spec.png_rrx
        ├── Concrete_Brown_Diff.png_rrx
        ├── Concrete_Norm.png_rrx
        ├── Concrete_Spec.png_rrx
        ├── Concrete_olive_Diff.png_rrx
        ├── Concrete_white_Diff.png_rrx
        ├── Foundation_Diff.png_rrx
        ├── Foundation_Norm.png_rrx
        ├── Foundation_Spec.png_rrx
        ├── Glass_Diff.png_rrx
        ├── Glass_Spec.png_rrx
        ├── Grass1_Diff.png
        ├── Grass1_Norm.png
        ├── Grass1_Spec.png
        ├── LaneMarking1_Diff.png
        ├── LaneMarking1_Norm.png
        ├── LaneMarking1_Spec.png
        ├── MetalWood_Trim_Diff.png_rrx
        ├── MetalWood_Trim_Norm.png_rrx
        ├── MetalWood_Trim_Spec.png_rrx
        ├── Roof_Bitumen_Diff.png_rrx
        ├── Roof_Bitumen_Norm.png_rrx
        ├── Roof_Bitumen_Spec.png_rrx
        ├── SanDiego_Building_Props_02_Diff.png_rrx
        ├── SanDiego_Building_Props_02_Norm.png_rrx
        ├── SanDiego_Building_Props_02_Spec.png_rrx
        ├── SandiegoProps_Diff.png_rrx
        ├── SandiegoProps_Norm.png_rrx
        ├── SandiegoProps_Spec.png_rrx
        ├── Stone_Trim_Diff.png_rrx
        ├── Stone_Trim_Norm.png_rrx
        ├── Stone_Trim_Spec.png_rrx
        ├── rr_random_props.fbx
        ├── rr_random_props.rrdata.xml
        └── rr_random_props.xodr

2) I run 
``` 
make import ARGS="--no-carla-materials"
```
which generates a map_package.json file in the Import directory, so my Import directory looks like this 

Import
└──map_package.json
└── custom_maps
    └── rr_random_props
        ├── Asphalt1_Diff.png
        ├── Asphalt1_Norm.png
        ├── Asphalt1_Spec.png
        ├── Brick1_Diff.png_rrx
        ├── Brick1_Norm.png_rrx
        ├── Brick1_Spec.png_rrx
        ├── Concrete1_Diff.png
        ├── Concrete1_Norm.png
        ├── Concrete1_Spec.png
        ├── Concrete3_Diff.png_rrx
        ├── Concrete3_Norm.png_rrx
        ├── Concrete3_Spec.png_rrx
        ├── Concrete_Brown_Diff.png_rrx
        ├── Concrete_Norm.png_rrx
        ├── Concrete_Spec.png_rrx
        ├── Concrete_olive_Diff.png_rrx
        ├── Concrete_white_Diff.png_rrx
        ├── Foundation_Diff.png_rrx
        ├── Foundation_Norm.png_rrx
        ├── Foundation_Spec.png_rrx
        ├── Glass_Diff.png_rrx
        ├── Glass_Spec.png_rrx
        ├── Grass1_Diff.png
        ├── Grass1_Norm.png
        ├── Grass1_Spec.png
        ├── LaneMarking1_Diff.png
        ├── LaneMarking1_Norm.png
        ├── LaneMarking1_Spec.png
        ├── MetalWood_Trim_Diff.png_rrx
        ├── MetalWood_Trim_Norm.png_rrx
        ├── MetalWood_Trim_Spec.png_rrx
        ├── Roof_Bitumen_Diff.png_rrx
        ├── Roof_Bitumen_Norm.png_rrx
        ├── Roof_Bitumen_Spec.png_rrx
        ├── SanDiego_Building_Props_02_Diff.png_rrx
        ├── SanDiego_Building_Props_02_Norm.png_rrx
        ├── SanDiego_Building_Props_02_Spec.png_rrx
        ├── SandiegoProps_Diff.png_rrx
        ├── SandiegoProps_Norm.png_rrx
        ├── SandiegoProps_Spec.png_rrx
        ├── Stone_Trim_Diff.png_rrx
        ├── Stone_Trim_Norm.png_rrx
        ├── Stone_Trim_Spec.png_rrx
        ├── rr_random_props.fbx
        ├── rr_random_props.rrdata.xml
        └── rr_random_props.xodr 

3) I then run 
```
make package ARGS="--packages=map_package"
``` 
which then creates a map_package_0.9.9-dirty.tar.gz in the Dist directory

4) I then move this tar file over to the Import directory of in the directory that contains the Carla binaries from where I'm going to start the Carla server, and run ./ImportAssets.sh in the Carla binaries directory to generate all the maps

5) I then start the server using 
```
SDL_VIDEODRIVER=offscreen SDL_HINT_CUDA_DEVICE=0 ./CarlaUE4.sh -ResX=800 -ResY=600 -nosound -windowed -opengl
```  

The reason I'm doing this is because eventually, I'm trying to come up with an automated way of importing multiple custom maps onto Carla, through RoadRunner/other ways. 

The error I'm seeing -  
When I visualise the map, I do not see the textures I have added on the buildings in RoadRunner in carla. I'm logging the outputs during both the make import and the make package phase. 

During the make import phase, there are a number of errors I'm seeing. One of them is the following - 
```
[2020.06.23-15.31.28:700][  0]LogFbxMaterialImport: Warning: Manual texture reimport and recompression may be needed for Concrete1_Norm
[2020.06.23-15.31.28:700][  0]LogFbxMaterialImport: Warning: Manual texture reimport and recompression may be needed for Concrete1_Norm
[2020.06.23-15.31.28:997][  0]LogFbxMaterialImport: Warning: Manual texture reimport and recompression may be needed for Concrete1_Norm
[2020.06.23-15.31.28:997][  0]LogFbxMaterialImport: Warning: Manual texture reimport and recompression may be needed for Concrete1_Norm
10011001100110011001100110011001100110011001100110011001[2020.06.23-15.31.29:520][  0]Error: Texture import failed
[2020.06.23-15.31.29:520][  0]Error: Texture import failed
10011001[2020.06.23-15.31.29:522][  0]Error: Texture import failed
[2020.06.23-15.31.29:522][  0]Error: Texture import failed
10011001[2020.06.23-15.31.29:524][  0]Error: Texture import failed
[2020.06.23-15.31.29:524][  0]Error: Texture import failed
10011001[2020.06.23-15.31.29:530][  0]Error: Texture import failed
[2020.06.23-15.31.29:530][  0]Error: Texture import failed
10011001[2020.06.23-15.31.29:531][  0]Error: Texture import failed
[2020.06.23-15.31.29:531][  0]Error: Texture import failed
10011001[2020.06.23-15.31.29:533][  0]Error: Texture import failed
[2020.06.23-15.31.29:533][  0]Error: Texture import failed
``` 
Right after the warning, I'm getting the "Texture Import failed" error. Why am I seeing this? What does it mean that "Manual texture reimport and recompression may be needed", and how do I resolve this?

bernatx, 2020-06-25:
Hi avishek-mondal,

Thanks for reporting this error.
It seems to me that you are referring to the wrong tuto for your pipeline. Because you want to import a whole map you should follow this tuto: [https://carla.readthedocs.io/en/latest/tuto_A_add_map/](url)

I think if you want to properly automatize the importing pipeline, you should export from Roadrunner using the option Export (no export to Carla) that will generate the FBX (with all mateirals required) and also the XODR file. Then you copy these files to carla/import/ and execute the **make import**, then the **make package ARGS="--packages=map_package"**.

If you are in Ubuntu, you can use the Docker that will make this process automatic and will output the package of the map already cooked. But if you are in windows, then you need to do this yourself with the Carla build from source.

My guess is that you exported for CARLA through RoadRunner option but then you don't import with the RoadRunner plugin, but executing the **make import** and probably the materials with extension **.png_rrx** are not valid for Unreal.

Let me know if that is the case. Try to follow the general export from RoadRunner and then the **make import** and see if that works. Otherwise, would be nice to have some files that we can check here.

Regards

avishek-mondal, 2020-06-25:
Hi! 
Sorry about that - yes I was referring to https://carla.readthedocs.io/en/latest/tuto_A_add_map
When I export from RoadRunner using just the standard fbx export (File->Export->Filmbox(.fbx)) 

[image]

, I still get a number of .png_rrx files. When I then generate the xodr file from RoadRunner, and then follow the same process as above with the **make import**, followed by **make package ARGS="--packages=map_package"**, I get the same Texture Import errors, and when I visualise the map on Carla, the textures are not there. 

I do agree with you that fundamentally the issue is that when I am trying to import without the RoadRunner plugin but use the .png_rrx files which is not being recognised by Unreal. For now, I'll find an alternative way to automating this pipeline, that perhaps either incorporates the RoadRunner plugin somehow, or uses some other method for creating the FBX and XODR files.

closed 2020-06-25 by avishek-mondal

rowandempster, 2020-09-18:
Hi @bernatx @avishek-mondal, I am having the same type of problem where using `make import` on a RoadRunner Carla export does not render correctly (e.g. no traffic lights or trees).
I am interested by @bernatx mentioning that 

> you exported for CARLA through RoadRunner option but then you don't import with the RoadRunner plugin, but executing the make import and probably the materials with extension .png_rrx are not valid for Unreal.

I was under the impression that using the `make import` was an _alternative_ to using the Unreal plugins, but it is the case that they have to be used in unison somehow? When I used the "deprecated" plugin method, my maps do render correctly, but then I can't export them as a standalone package. Is there a way to use the plugins (so that rendering is correct) and also export to a standalone plugin in the same pipeline?

Thanks!

bernatx, 2021-05-03:
Hi @jrefi, Road Runner is not needed for importing maps but only recommended because it makes the process of creating a map easier. But whatever you want to import, you need to have a .FBX file with the geometry of the map (also materials you want to use) and a .XODR (openDRIVE) with all the road network information. That is not related to Road Runner, any other software that create those files is fine, but you need those files in order to import a map. 
We know there are some issues/bugs on how Unreal import the FBX, but in overall it works fine.
With those files you only need to put into the **import** folder and let the .JSON be created automatically in the process.
And that should be enough.

Anoter point to take care is the nomenclature, because we use them to indicate to our pedestrians where they can walk and where they can't.

FBanani, 2023-06-28:
> Hi @jrefi, Road Runner is not needed for importing maps but only recommended because it makes the process of creating a map easier. But whatever you want to import, you need to have a .FBX file with [...]

Hi,

I have an issue when importing large map. I do not see any error but in the unreal there is nothing there!
[image]