carla-simulator/carla #2988: build error: CXX object LibCarla/cmake/server/CMakeFiles/c....dir/__/__/source/carla/opendrive/parser/GeometryParser.cpp. FAILED:

gengtsh, 2020-06-23:
hi
when i make launch, i got the 
Building CXX object LibCarla/cmake/server/CMakeFiles/carla_server.dir/__/__/source/carla/road/element/Geometry.cpp.o
FAILED:
Building CXX object LibCarla/cmake/server/CMakeFiles/c...rver.dir/__/__/source/carla/opendrive/parser/RoadParser.cpp.
FAILED:
Building CXX object LibCarla/cmake/server/CMakeFiles/c....dir/__/__/source/carla/opendrive/parser/GeometryParser.cpp.
FAILED: 

PS: i use clang++-10, ubuntu 20.04
i try to install clang++-8 on Ubuntu 20;04, but failed

any help are welcome

germanros1987, 2020-06-25:
@Axel1092 could you please look into this?

Axel1092, 2020-07-03:
Hi @gengtsh,
These errors are probably due to using clan10 instead of clang8. What problems did you find with installing clang8?
Unreal Engine is compiled with this version of clang and mixing version may produce errors. I suggest to try to find a way to use clang8.

gengtsh, 2020-07-03:
hi, @Axel1092,
i remove everything related to clang10 and reinstall clang8, it looks like
this issues is fixed, but i encounter other issues: see the links below:
https://github.com/carla-simulator/carla/issues/3020
https://github.com/carla-simulator/carla/issues/3021

closed 2020-09-11 by stale[bot]