carla-simulator/carla #2989: Get current RPM of vehicle

raymanfx, 2020-06-23:
Hi, I'm trying to find a way to get the current revolutions per minute (RPM) of any wheel at a given time via the Python API.
I know I can get the throttle level and the max RPM of the engine via `VehiclePhysicsControl`, so in theory I could just derive the wheel RPM from there. Due to gearbox properties etc. that value will be pretty inaccurate I guess.

What I'm actually more worried about there is physics simulation. If the car goes uphill, the actual wheel RPM will obivously not scale in the same way with the engine RPM anymore. Other cases would be increased/lowered road friction, e.g. on an icy road.

TL;DR is there a way to access wheel RPM via the Python API?

germanros1987, 2020-06-25:
I think we don't have a simple way to do that in our current API. We should add it. 

@bernatx do you think we could find a simple workaround for this? Otherwise, let's add it to the backlog.

raymanfx, 2020-07-01:
I didn't find an easy workaround. Thank you for adding the feature to the backlog!
As an intermediate step, I am interested in engine RPM, but could not find an easy API to get it. Do you have a quick formula to derive the engine RPM from the gear ratios etc?

germanros1987, 2020-07-02:
@marcgpuig do you have any ideas?

marcgpuig, 2020-07-03:
@raymanfx 
I don't know if UE4 internally uses this information. If it's there it can be exposed but further investigation is needed.
If you find something make sure to report it here.

raymanfx, 2020-07-03:
@marcgpuig this is my dummy solution that I'm using for demo purposes:
```
        c = vehicle.get_control()
        p = vehicle.get_physics_control()

        # TODO: proper simulation
        # It is still unclear to me how the gears are modeled in CARLA:
        #   -1 reverse
        #   0  neutral
        # But how does this map to the gear ratios?
        engine_rpm = p.max_rpm * c.throttle
        if c.gear > 0:
            gear = p.forward_gears[c.gear]
            engine_rpm *= gear.ratio
```
This actually appears to yield quite reasonable engine RPM values, but it's of course lacking much of the underlying physics simulation necessary to get the actual engine RPM.

marcgpuig, 2020-07-06:
@raymanfx thanks! Hopefully, this can help someone else. In the meantime, we will continue to push for this feature at some point.

raymanfx, 2020-09-05:
This is still relevant to me. I see you assigned this to milestone 0.9.11. Is there any progress so far? My team will start to evaluate options regarding RPM values, but we are not as familiar with CARLA internals as you are of course.

closed 2020-11-14 by stale[bot]

kmeids, 2021-05-18:
Hello Everyone,
I don't seem to find this feature in the 9.11 release, is it still on the backlog?
I can confirm that having this feature will help support several use-cases especially when coupling physics-based radar simulation to CARLA.

raymanfx, 2021-05-18:
FYI, Unreal Engine 4.24 (used by Carla 0.9.11) makes use of PhysX for vehicle physics.
Unreal Engine 4.26 (which will be used by Carla 0.9.12) ships their own Chaos physics engine for vehicles, which is meant to replace PhysX.

To expose wheel RPM in Carla, we will have to make it available from the Unreal Engine code first. My team is looking into this right now, but it's on low priority.

AOOOOOA, 2023-03-24:
> FYI, Unreal Engine 4.24 (used by Carla 0.9.11) makes use of PhysX for vehicle physics. Unreal Engine 4.26 (which will be used by Carla 0.9.12) ships their own Chaos physics engine for vehicles, [...]

Hi Have you finished this function yet?

YuhengZhi, 2024-08-26:
> @marcgpuig this is my dummy solution that I'm using for demo purposes:
>
> ```
>         c = vehicle.get_control()
>         p = vehicle.get_physics_control()
>
>         # TODO: proper simulation
>         # [...]

I bumped into this thread when trying to investigate the exact mechanism of vehicle dynamics in CARLA with respect to the control inputs, especially the normalized `throttle`. After some investigation I don't think this function would give you the current engine RPM as `throttle` scales the max torque at the current engine RPM, not RPM itself. See [these lines](https://github.com/CarlaUnreal/UnrealEngine/blob/e8c73304f9f3f91ba12c997516deeb2434e4489e/Engine/Source/ThirdParty/PhysX3/PhysX_3.4/Source/PhysXVehicle/src/PxVehicleUpdate.cpp#L1424-L1428) from PhysX_3.4 (requires linking unreal engine account to github):
```
PX_FORCE_INLINE PxF32 computeEngineDriveTorque(const PxVehicleEngineData& engineData, const PxF32 omega, const PxF32 accel)
{
	const PxF32 engineDriveTorque=accel*engineData.mPeakTorque*engineData.mTorqueCurve.getYVal(omega*engineData.getRecipMaxOmega());
	return engineDriveTorque;
}
```
The `accel` value here is almost equal to `throttle` up to some smoothing mechanism. 

I believe a better way to estimate wheel RPM would be to divide the linear velocity of wheel at the contact point with road surface by the radius of the wheel, but I am not sure though.