carla-simulator/carla #2992: Waypoint API calls are very expensive

hh0rva1h, 2020-06-25:
We are doing reinforcement learning and want to track the offset of the car from the lane center. Therefore we make use of the waypoint API, however we noticed that that get_waypoint() calls affect performance very badly.

We wrote a script to outline this performance issue:
https://gist.github.com/hh0rva1h/ff708da80e6e7c61c4e41894ddcbb198

On my machine I obtain the following results:
```
python3 benchmark_waypoints.py 
performance: 609.0 ticks per second for 1.0 seconds
performance (waypoints):   8.0 ticks per second for 1.0 seconds
```
However we noticed that the lane invasion sensor does not drop performance (so it seems internally the lane invasion sensor does not make use of waypoints or it does it more efficiently).

So could you please evaluate whether the performance of those calls could be improved? Or is this to be expected and we should use other means to track the offset of the car, e.g. receive all waypoints in one call and estimate the nearest way-point ourselves in every time step ourselves?

@germanros1987 Could you be so kind to triage this issue?
@marcgpuig Since you implemented the API could you maybe leave a comment on whether improving the performance of those calls is sensible? Or if it would make sense to make it possible to register a callback/sensor that could report the nearest waypoint with higher performance?

marcgpuig, 2020-07-03:
Hi @hh0rva1h, sorry for the late response!

With your same script and the current `master` branch in my machine (_2080 ti_ | _Intel(R) Xeon(R) CPU E5-1650 v4 @ 3.60GHz_):
```
performance:  60.0 ticks per second for 1.0 seconds
performance (waypoints):  11.0 ticks per second for 1.0 seconds
```
The main problem is that you are using `world.get_map()` on every tick, and this basically ask for the OpenDRIVE file to the server, retrieves it as a string, and parses it to get the actual `carla.map` object. Considering that the `Town04.xodr` is around 54 MB I think 8-11 ticks per second still very fast :smile:

You should do:
```py
...

client = carla.Client("localhost", 2000)
world = client.load_world("Town04")
m = world.get_map()

...

while time.perf_counter() - t0 < benchmark_time:
    world.tick()
    m.get_waypoint(vehicle.get_location())

...
```
Then, I get the following:
```
performance:  60.0 ticks per second for 1.0 seconds
performance (waypoints):  60.0 ticks per second for 1.0 seconds
```
Also, notice that these queries are not sent to the simulator, these are computed in the client itself, so synchronizing with the server is pointless if you are benchmarking the Waypoint API. In fact, you can create a script that does not connect to the simulator, and just loads an OpenDRIVE file to make all of these available Waypoint queries to it, which is very useful to make all sorts of utilities. If you need some examples, I can share here some tools I made.

Finally, let me recommend you to store an initial settings state when working with sync, so when the script is done you can set the simulator to async again so it won't get stuck.
```py
...

init_settings = world.get_settings()
settings = world.get_settings()
settings.fixed_delta_seconds = 0.05
settings.synchronous_mode = True

...

world.apply_settings(init_settings)
world.tick()
```
Hope this helps, cheers!

marcgpuig, 2020-07-03:
Let me ping @sergi-e so this knowledge can be added to our documentation in `carla.World.get_map()` and people don't get confused about this. Thanks for your report.

hh0rva1h, 2020-09-02:
@marcgpuig @sergi-e Thanks very much, has this been added to the doc?

wendersonj, 2020-10-28:
I don't know if it's relevant, but when using World objet and some other function that need waypoints, don't use the call `carla.get_map().get_wapoint()`. Create a variable `map` and use `map.get_waypoint()`

It's a lot faster and helpful.
Cheers!

PR #3356 (2020-09-24): sergi-e/docs-map-issue-2992
#### Description

- Some notes and clearer documentation regarding the waypoint API.
In summary, the world.get_map() method does call the server, but the waypoint API itself happens on the client-side. 

#### Fixes

#2992 

#### Where has this been tested?

  * **Platform(s):** Ubuntu 18.04.4
  * **Python version(s):** 3.6.9
  * **Unreal Engine version(s):** 4.24

---
This change is [[image]](https://reviewable.io/reviews/carla-simulator/carla/3356)
closed 2021-01-26 by germanros1987