carla-simulator/carla #2994: Unable to control vehicle in Carla with Logitech G29 steering wheel

elandg, 2020-06-25:
I've been trying to run Carla using the manual_control_steeringwheel.py file, but am repeatedly getting "NoSectionError: No section: 'G29 Racing Wheel'". I will attach the full terminal output below. I am using the 0.9.9.2 Windows version of Carla, starting CarlaUE4.exe first, and using the Logitech G29 steering wheel. After reading some other posts, I have included a wheel_config.ini file in the carla folder, installed Logitech G Hub, and have tried using `self._parser.read(r"C:/Users/Eric Landgraf/Default/Umich/UMTRI/CARLA_0.9.9.2-W/PythonAPI/examples/wheel_config.ini")`, which should be the absolute path (both will be attached as text files). If anyone has gotten their Logitech G29 steering wheel to work with Carla, please let me know what I should change.

[image]

[wheel_config.txt](https://github.com/carla-simulator/carla/files/4833045/wheel_config.txt)

[manual_control_steeringwheel.txt](https://github.com/carla-simulator/carla/files/4833041/manual_control_steeringwheel.txt)

elandg, 2020-07-01:
I apologize for the simple mistake, I had moved the "wheel_config.ini" file to the main carla folder without changing the absolute path. However, I am now getting an out of bounds error. 
[image]
I have uncommented out the line `print (jsInputs)` and that has lead to the [] you see printed above the error message. I am going to try reconfiguring my g29 steering wheel and see if that helps. Any other suggestions are also appreciated.

closed 2020-09-06 by stale[bot]

will-sloan, 2020-09-19:
@elandg Did you figure out the list index out of range?

chengchihhong, 2020-10-21:
> @elandg Did you figure out the list index out of range?

This is because the number of axes is 0, making the value in jsInputs to be wrong. If you try self._joystick.get_numaxes(), you can see it. 

To check if it is possible to do it in Win10, I have tried some workarounds but there seems to be no simple fix.

- I tried to replace pygame by pygame_sdl2. This helps in detecting 4 axes.
- Then there are more problems - as pygame_sdl2 does not implement everything, so many warnings are out. 
- I stopped pushing when I realized that I need to also rewrite pygame.surfarray.make_surface(), as this is not supported by pygame_sdl2.

will-sloan, 2020-10-22:
@chengchihhong 
Turns out pygame has a branch that has sdl2. Using that solved the problem.

seuwcs, 2021-04-09:
@elandg  Hello, if you have the problem solved?

Shroculus, 2022-02-03:
Hi,

Could someone please let me know if this bug was fixed? because I am facing the exact same error with 0.9.13 Carla when trying to calibrate it with G920 steering wheel. also, could you direct me in the right direction if this has already been solved?

Thanks in advance.