carla-simulator/carla #2996: Add noise to the autopilot control

magehrig, 2020-06-26:
I want to add noise to the steering of the autopilot on every tick of the simulation.

**What I tried**: Add a callback via the` carla.World.on_tick` method as follows:

```Python
def my_callback(vehicle, world_snapshot):
    control = vehicle.get_control()
    control.steer += random.uniform(-0.1, 0.1)
    vehicle.apply_control(control)

world.on_tick(functools.partial(add_steer_noise, vehicle))
```

Unfortunately, this had no effect on the steering control. Is there any other way to achieve this with the Python API?

I am using the precompiled version 0.9.9.4.

This question was already asked in https://github.com/carla-simulator/carla/issues/1522 but the provided answers did not work for me either, and I cannot reopen issues.

jackbart94, 2020-06-26:
Hi,

You should add it directly on the Traffic Manager API. If you go to ``MotionPlanStage.cpp`` and look for 

```// Constructing the actuation signal.
  if (ego_physics_enabled) {
    carla::rpc::VehicleControl vehicle_control;
    vehicle_control.throttle = actuation_signal.throttle;
    vehicle_control.brake = actuation_signal.brake;
    vehicle_control.steer = actuation_signal.steer;

    output_array.at(index) = carla::rpc::Command::ApplyVehicleControl(actor_id, vehicle_control);
  }
```
You can add the random.uniform(-0.1, 0.1) to the actuation_signal before the vehicle_control is applied.

magehrig, 2020-06-26:
Thanks I will give it a shot and report back

germanros1987, 2020-06-27:
Hi @magehrig 

So it seems I was wrong and you can't inject this noise directly in Python (yet). In any case, the change seems small. We could help out by enabling/updating the interface to support noise.

Let us know if you need help with this or if you are just implementing this by yourself.

magehrig, 2020-06-30:
@jackbart94 I recompiled CARLA with your suggestions. Unfortunately it does not seem to make a difference either. This is a 10 second sequence with the proposed changes using autopilot: https://youtu.be/_tOPd97Hchc.

magehrig, 2020-06-30:
@jackbart94 I recompiled CARLA with your suggestions. Unfortunately it does not seem to make a difference either. This is a 10 second sequence with the proposed changes using autopilot: https://youtu.be/_tOPd97Hchc.

The code I used for this is:
```C++
// Constructing the actuation signal.
if (ego_physics_enabled) {

std::default_random_engine generator;
std::uniform_real_distribution<double> distribution(-0.1,0.1);
double number = distribution(generator);

carla::rpc::VehicleControl vehicle_control;
vehicle_control.throttle = actuation_signal.throttle;
vehicle_control.brake = actuation_signal.brake;
vehicle_control.steer = actuation_signal.steer + static_cast<float>(number);

output_array.at(index) = carla::rpc::Command::ApplyVehicleControl(actor_id, vehicle_control);
} else {
```
Is the steering in radians/second or degrees/second btw?

jackbart94, 2020-06-30:
Hi,

The code isn't correct. By printing the random number, you can see it's always the same. We are actually implementing a random generation class, soon it will be in master, but in the meantime you can check out how to do it in the branch ``traffic_manager/determinism_check``.

magehrig, 2020-07-06:
I used your `traffic_manager/determinism_check` branch and implemented the steering noise using the RandomGenerator class. However, I can still not produce the desired effect. I don't have time to investigate this issue further but we should keep it open unless it is solved (don't let it go stale).

magehrig, 2020-07-06:
Thanks I could make it work using the `RandomGenerator` class of your branch. Although, it seems that it had an effect on all the cars not just the own vehicle that I wanted to steer. Is this to be expected?

jackbart94, 2020-07-08:
Yeah, that's definitely to be expected. You are assigning this random noise steering behavior to every vehicle that is controlled by the Traffic Manager or autopilot. If you want to assign it only to your vehicle (or any subset of all the vehicles controlled by the TM) you have to implement a parameter, such as we do with ignore_traffic_light_percentage, and then in your script only assign this particular behavior to your vehicle.

closed 2020-07-30 by germanros1987

wu0023, 2022-08-30:
Hi, I would like to know how to call the Randomgenerator from Python API (I had no experience on the C++), and whether it is possible to add custom noise to the control command in autopilot mode.