carla-simulator/carla #2999: Timeout parameter for spawn_npc.py

faultfactory, 2020-06-26:
I run in synchronous mode and I develop carla scripts and such using a lower power computer, review the results and then run on a more powerful machine when it becomes available. I have been git stash / applying a change to allow a timeout parameter to be passed for the wait_for_tick so it is above the default of 10.0s. 

I've put this into a branch and can push it. It would be good to have it as part of the repo. 
I did not see anything related to the documentation that requires modification for this update but I may have missed it.

faultfactory, 2020-06-26:
Also, i attempted to push this branch but i assume i need to be added as a contributor first. It is a small change.

jackbart94, 2020-06-26:
Hi,

You can open a PR for this, but I'm pretty sure that wait_for_tick already allows you to input your own value.

faultfactory, 2020-06-26:
It does, but you cannot do it from the command line

faultfactory, 2020-06-26:
Do i have to do this from my own fork or can i push a branch to this repo? First PR for me on github.

JMorceaux, 2020-06-26:
Well, not sure if i'll answer your question, but If you read the doc, you can see the following:
```
wait_for_tick(self, seconds=10.0)
   - The client tells the server to block calling thread until a world_tick() is received.
          Parameters:
                 seconds (float) – Maximum time in seconds the server should wait for a tick. 
                                             It is set to 10.0 by default.
          Return: carla.WorldSnapshot
```
So I guess, it is already implemented

faultfactory, 2020-06-26:
I want to pass this timeout to the spawn_npc.py from the command line. Essentially i'm adding another input argument to the spawn_npy.py script and then passing that argument to the wait_for_tick.  Since i run the same set of code on two machines with different capabilities and speeds. Instead of modifying the scripts, I want to set it up so i can change this timeout parameters from the command line.

germanros1987, 2020-06-27:
Hi @faultfactory,

Thanks for reaching out. Sure, just make a small PR to update spawn_npc. We will review it, and if we don't find possible conflicts we will merge it. 

Please, remember to ping this issue in your PR.

Thanks!

faultfactory, 2020-06-29:
@germanros1987 

Awesome, thanks for getting back to me. 

What is the proper process for pushing a community PR to the repository? I am attempting to push a new branch on the command line but i'm getting the 403 forbidden error.

germanros1987, 2020-06-29:
All good now.

closed 2020-08-31 by jackbart94