carla-simulator/carla #3003: cann't listen the camera data.

HuangSecretKey, 2020-06-29:
I use the sensor.listen(）and cv2.imshow（) to show the image，but the window didn't show video and only the first image.And the error is “QWaitCondition: cv destroy failure: device or resource busy
QWaitCondition: mutex destroy failure: device or resource busy”Also，when I set the sensor_tick be a bigger value, this will disappear. I don't know the reason. Can anyone help me? The version is 0.9.9.8
the code is as follows：
```
def process_img(image):
i = np.array(image.raw_data)
i2 = i.reshape((IM_HEIGHT, IM_WIDTH, 4))
i3 = i2[:,:,:3]
cv.imshow(WINDOW_NAME,i3)
cv.waitKey(1)
return i3/255.0
sensor.listen(lambda data: process_img(data))
```

germanros1987, 2020-06-30:
Hi @HuangSecretKey . Please, could you double-check the version of CARLA? 0.9.9.8 doesn't exist yet. Actually, I doubt we reach the 8th review of 0.9.9 before releasing 0.9.9

In any case, those errors seem to be related to OpenCV, not to CARLA. So this is probably not the right place to ask this question. You could probably try through [Discord](https://discord.gg/8kqACuC) or the [forum](forum.carla.org).

Best regards.

germanros1987, 2020-07-30:
Issue closed due to lack of activity. Feel free to open a new issue if you still consider this is a critical matter.

closed 2020-07-30 by germanros1987