carla-simulator/carla #3010: Coordinates of actor

Vin1291, 2020-06-30:
Carla's client bounding box gives coordinates in the camera frame. How can we get the coordinates of actors in the world frame? can anyone suggest to me how can we get coordinates of actors in the world frame instead of camera frame?

germanros1987, 2020-07-02:
@marcgpuig could you provide an example, please?

bender2po, 2020-07-03:
> Carla's client bounding box gives coordinates in the camera frame. How can we get the coordinates of actors in the world frame? can anyone suggest to me how can we get coordinates of actors in the world frame instead of camera frame?

If you follow the file 'client_bounding_boxes.py', you can calculate it. 
See below snippet for example ( copied the lines from 'client_bounding_boxes.py'):

```
def get_matrix(transform):

        rotation = transform.rotation
        location = transform.location
        c_y = np.cos(np.radians(rotation.yaw))
        s_y = np.sin(np.radians(rotation.yaw))
        c_r = np.cos(np.radians(rotation.roll))
        s_r = np.sin(np.radians(rotation.roll))
        c_p = np.cos(np.radians(rotation.pitch))
        s_p = np.sin(np.radians(rotation.pitch))
        matrix = np.matrix(np.identity(4))
        matrix[0, 3] = location.x
        matrix[1, 3] = location.y
        matrix[2, 3] = location.z
        matrix[0, 0] = c_p * c_y
        matrix[0, 1] = c_y * s_p * s_r - s_y * c_r
        matrix[0, 2] = -c_y * s_p * c_r - s_y * s_r
        matrix[1, 0] = s_y * c_p
        matrix[1, 1] = s_y * s_p * s_r + c_y * c_r
        matrix[1, 2] = -s_y * s_p * c_r + c_y * s_r
        matrix[2, 0] = s_p
        matrix[2, 1] = -c_p * s_r
        matrix[2, 2] = c_p * c_r
        return matrix
    
def _vehicle_to_world(cords, vehicle):

    bb_transform = carla.Transform(vehicle.bounding_box.location)
    bb_vehicle_matrix = get_matrix(bb_transform)
    vehicle_world_matrix = get_matrix(vehicle.get_transform())
    bb_world_matrix = np.dot(vehicle_world_matrix, bb_vehicle_matrix)
    world_cords = np.dot(bb_world_matrix, np.transpose(cords))
    return world_cords

def _create_bb_points(vehicle):

    cords = np.zeros((8, 4))
    extent = vehicle.bounding_box.extent
    cords[0, :] = np.array([extent.x, extent.y, -extent.z, 1])
    cords[1, :] = np.array([-extent.x, extent.y, -extent.z, 1])
    cords[2, :] = np.array([-extent.x, -extent.y, -extent.z, 1])
    cords[3, :] = np.array([extent.x, -extent.y, -extent.z, 1])
    cords[4, :] = np.array([extent.x, extent.y, extent.z, 1])
    cords[5, :] = np.array([-extent.x, extent.y, extent.z, 1])
    cords[6, :] = np.array([-extent.x, -extent.y, extent.z, 1])
    cords[7, :] = np.array([extent.x, -extent.y, extent.z, 1])
    return cords

bb_cords = _create_bb_points(vehicle)
world_cord = _vehicle_to_world(bb_cords, vehicle)
```

Vin1291, 2020-07-03:
@bender2po Yeah, I figured that out and then how do you get such coordinates of actors in reference to the main ego vehicle?

bender2po, 2020-07-03:
> @bender2po Yeah, I figured that out and then how do you get such coordinates of actors in reference to the main ego vehicle?

It is simple.
You can calculate 'world_cord' from my reply above. Then, use '_world_to_sensor(world_cords, ego_vehicle)'.

marcgpuig, 2020-07-03:
@bender2po we just added #2997.
Finally, there is no need to compute this matrix yourself :tada:

Vin1291, 2020-07-03:
@marcgpuig Thank you for your reply, can you please explain do we use this feature and calculates bounding box coordinates of all the actors with respect to ego vehicle. I got confused a bit.

Vin1291, 2020-07-03:
> > @bender2po Yeah, I figured that out and then how do you get such coordinates of actors in reference to the main ego vehicle?
> 
> It is simple.
> You can calculate 'world_cord' from my reply above. Then, use '_world_to_sensor(world_cords, ego_vehicle)'.

If I do this, then I can't see bounding boxes on the actors.

Vin1291, 2020-07-03:
@bender2po I did this and getting such bounding box, as you can they are shifted.

    @staticmethod
    def get_bounding_box(vehicle, camera, ego_vehicle):
        """
        Returns 3D bounding box for a vehicle based on camera view.
        """

        bb_cords = ClientSideBoundingBoxes._create_bb_points(vehicle)
        cords_x_y_z = ClientSideBoundingBoxes._vehicle_to_sensor(bb_cords, vehicle, camera, ego_vehicle)[:3, :]
        cords_y_minus_z_x = np.concatenate([cords_x_y_z[1, :], -cords_x_y_z[2, :], cords_x_y_z[0, :]])
        bbox = np.transpose(np.dot(camera.calibration, cords_y_minus_z_x))
        camera_bbox = np.concatenate([bbox[:, 0] / bbox[:, 2], bbox[:, 1] / bbox[:, 2], bbox[:, 2]], axis=1)
        return camera_bbox

    @staticmethod
    def _create_bb_points(vehicle):
        """
        Returns 3D bounding box for a vehicle.
        """

        cords = np.zeros((8, 4))
        extent = vehicle.bounding_box.extent
        cords[0, :] = np.array([extent.x, extent.y, -extent.z, 1])
        cords[1, :] = np.array([-extent.x, extent.y, -extent.z, 1])
        cords[2, :] = np.array([-extent.x, -extent.y, -extent.z, 1])
        cords[3, :] = np.array([extent.x, -extent.y, -extent.z, 1])
        cords[4, :] = np.array([extent.x, extent.y, extent.z, 1])
        cords[5, :] = np.array([-extent.x, extent.y, extent.z, 1])
        cords[6, :] = np.array([-extent.x, -extent.y, extent.z, 1])
        cords[7, :] = np.array([extent.x, -extent.y, extent.z, 1])
        return cords

    @staticmethod
    def _vehicle_to_sensor(cords, vehicle, sensor, ego_vehicle):
        """
        Transforms coordinates of a vehicle bounding box to sensor.
        """

        world_cord = ClientSideBoundingBoxes._vehicle_to_world(cords, vehicle)
        sensor_cord = ClientSideBoundingBoxes._world_to_sensor(world_cord, ego_vehicle)
        return sensor_cord

    @staticmethod
    def _vehicle_to_world(cords, vehicle):
        """
        Transforms coordinates of a vehicle bounding box to world.
        """

        bb_transform = carla.Transform(vehicle.bounding_box.location)
        bb_vehicle_matrix = ClientSideBoundingBoxes.get_matrix(bb_transform)
        vehicle_world_matrix = ClientSideBoundingBoxes.get_matrix(vehicle.get_transform())
        bb_world_matrix = np.dot(vehicle_world_matrix, bb_vehicle_matrix)
        world_cords = np.dot(bb_world_matrix, np.transpose(cords))
        return world_cords

    @staticmethod
    def _world_to_sensor(cords, sensor):
        """
        Transforms world coordinates to sensor.
        """

        sensor_world_matrix = ClientSideBoundingBoxes.get_matrix(sensor.get_transform())
        world_sensor_matrix = np.linalg.inv(sensor_world_matrix)
        sensor_cords = np.dot(world_sensor_matrix, cords)
        return sensor_cords

    @staticmethod
    def get_matrix(transform):
        """
        Creates matrix from carla transform.
        """

        rotation = transform.rotation
        location = transform.location
        c_y = np.cos(np.radians(rotation.yaw))
        s_y = np.sin(np.radians(rotation.yaw))
        c_r = np.cos(np.radians(rotation.roll))
        s_r = np.sin(np.radians(rotation.roll))
        c_p = np.cos(np.radians(rotation.pitch))
        s_p = np.sin(np.radians(rotation.pitch))
        matrix = np.matrix(np.identity(4))
        matrix[0, 3] = location.x
        matrix[1, 3] = location.y
        matrix[2, 3] = location.z
        matrix[0, 0] = c_p * c_y
        matrix[0, 1] = c_y * s_p * s_r - s_y * c_r
        matrix[0, 2] = -c_y * s_p * c_r - s_y * s_r
        matrix[1, 0] = s_y * c_p
        matrix[1, 1] = s_y * s_p * s_r + c_y * c_r
        matrix[1, 2] = -s_y * s_p * c_r + c_y * s_r
        matrix[2, 0] = s_p
        matrix[2, 1] = -c_p * s_r
        matrix[2, 2] = c_p * c_r
        return matrix
[image]

bender2po, 2020-07-03:
@Vin1291 I need more information to answer your question. How did you capture the image which you have shared in the above post? (1) is it from server window? or (2) is it from image from camera-rgb sensor?
If it is the (1) then it should work.
if it is the (2). then it is expected to get a picture like you have shared. So, if you are trying to project on to the image captured from camera-sensor then you shall do as exactly in the 'client_bounding_boxes.py'

Vin1291, 2020-07-05:
I am capturing the image from camera -rgb sensor, but I don't get it why is it showing like that. Can you tell me how to fix it?

closed 2020-09-12 by stale[bot]