carla-simulator/carla #3011: Docker build issue

sourabhmisal, 2020-07-01:
I am facing this issue in ubuntu 18.04, when I run `docker build -t carla -f Carla.Dockerfile .`

`Parsing headers for CarlaUE4Editor Running UnrealHeaderTool "/home/ue4/carla/Unreal/CarlaUE4/CarlaUE4.uproject" "/home/ue4/carla/Unreal/CarlaUE4/Intermediate/Build/Linux/B4D820EA/CarlaUE4Editor/Development/CarlaUE4Editor.uhtmanifest" -LogCmds="loginit warning, logexit warning, logdatabase error" -Unattended -WarningsAsErrors Error: Couldn't find parent type for 'CarlaLightSubsystem' named 'UWorldSubsystem' in current module or any other module parsed so far. make[1]: *** [CarlaUE4Editor] Error 5 Makefile:215: recipe for target 'CarlaUE4Editor' failed make[1]: Leaving directory '/home/ue4/carla/Unreal/CarlaUE4' make: *** [CarlaUE4Editor] Error 2 Util/BuildTools/Linux.mk:79: recipe for target 'CarlaUE4Editor' failed The command '/bin/sh -c cd /home/ue4 && if [ -z ${GIT_BRANCH+x} ]; then git clone --depth 1 https://github.com/carla-simulator/carla.git; else git clone --depth 1 --branch $GIT_BRANCH https://github.com/carla-simulator/carla.git; fi && cd /home/ue4/carla && ./Update.sh && make CarlaUE4Editor && make PythonAPI && make build.utils && make package && rm -r /home/ue4/carla/Dist' returned a non-zero code: 2
[image]
`

Anyone knows the solution for this?

germanros1987, 2020-07-02:
@bernatx any idea of what could be happening here?

albinnyASTAzero, 2020-07-03:
I had the same issue but I managed to go around it by building the supported docker version of UE4 for the Carla version I'm using. You can check in the Prerequisites.Dockerfile at the top which one it should be. I used the latest release of Carla which at the top of the file it says ARG UE4_V=4.24.3. So I built that version ( ue4-docker build 4.24.3 --no-engine --no-minimal) instead of 4.22.2 which is stated in the guide your using and it worked fine.

bernatx, 2020-07-06:
We will update that file so it says 4.24 instead of 4.22 to avoid confusions.
Thanks for the warning.
It solved your problem @sourabhmisal ?

sourabhmisal, 2020-07-06:
Yes , it did , thanks for the support!

closed 2020-07-06 by sourabhmisal