carla-simulator/carla #3012: Carla 0.9.9 crashing way more than previous version

marintoro, 2020-07-01:
Hello, I am doing Reinforcement Learning in CARLA for more than a year and I have always encountered issues with CARLA crashing time to time (with a Segmentation Fault as mentionned in #1627 or #1227). But generally this was happening really rarely. Usually I have 5/6 instance of CARLA running independently during around 4/5 days and sometimes one of this instance of CARLA would crash and I had to relaunch it.

I recently moved from CARLA 0.9.6 to CARLA 0.9.9 and I discovered that the crashs were happening much more, basically I have trouble to run a CARLA for more than a day without crash. Particularly my script (a slight modification to the spawn_npc.py script) spawning/deleting/respawning all the surrounding autopilot vehicles and pedestrians seems really unstable throwing this exception many time:
***$ what(): length > 2.0f * std::numeric_limits<float>::epsilon()***
***Segmentation fault (core dumped)***

As in the script spawn_npc.py I set the traffic manager with those commands:
***
$ traffic_manager.set_global_distance_to_leading_vehicle(2.0)
$ traffic_manager.set_hybrid_physics_mode(True)
***

jackbart94, 2020-07-01:
Hi,

Sorry for this issue! Could you provide us some extra information? If possible, running some debugger (like gdb). Or maybe the script to test this?

marintoro, 2020-07-01:
Hello,

I just made a short version of my script to delete and respawn vehicle and pedestrian every 10 seconds and after just some minutes, CARLA crashs.

I am launching CARLA with this command line:
$ ./CarlaUE4.sh -quality-level=Epic -world-port=2000 -resx=100 -resy=100

germanros1987, 2020-07-02:
@jackbart94 please give high priority to this issue.
@bernatx keep an eye on this.

jackbart94, 2020-07-02:
Hi,

I've been running it for about 30 minutes and can't seem to get any sort of error. I'm on Ubuntu 16.04 with Python 3.7.2 and CARLA 0.9.9.4. Maybe you could try getting this version of CARLA and let me know if you still experience it, or as I said send me the report by using the gdb debugger.

marintoro, 2020-07-02:
Hi,

Could you detail a bit more what you mean by "using the gdb debugger"?

And this kind or error is hard to detect/reproduce, sometimes it doesn't crash for multiple hours, sometimes it crashs after just 20/30 minutes. 
When I am launching my whole reinforcement learning training process, CARLA usually crash after about 20 hours.

But with my minimal script if I let it 2 minutes, then kill and restart it 3 or 4 times every 2 minutes I consistently reach this kind of crash after only 10/15 minutes which reveal in my opinion the non-stability of the process.
[image]

jackbart94, 2020-07-02:
``Could you detail a bit more what you mean by "using the gdb debugger"?``

Sure, please run in the following order:
1. ``` $ ./CarlaUE4.sh -quality-level=Epic -world-port=2000 -resx=100 -resy=100```
2. (in another terminal) ```gdb python```
3. (in the same terminal as command number 2) ```run rl/rainbow_ign_apex/issue_carla.py```
Then, once it crashes, in the same terminal as command number 2 just write ```bt``` and it will print the stack trace of the error.

Also, I've been running it and still it won't crash.. and your seg fault in this screenshot doesn't show the error you mentioned.

marintoro, 2020-07-02:
For exemple, this time I got the issue after only 30 seconds...
[image]

And yes this is not the exact same error... When I am launching the RL training I get the error I mentioned (but I don't want to run it for multiple days, I just wanted to give one exemple of possible crash...).

With my minimal script however, I succeed to get a Segmentation Fault consistently and I assume this is related to the same "unstability" problems.

My main question actually is why this doesn't give any Python exception? Like this we could catch each time there is a crash and restart everything automatically and properly. In the current state I just can't do anything from Python side...

jackbart94, 2020-07-02:
Did you try switching to CARLA 0.9.9.4 by any chance? Or are you still on 0.9.9? 

> My main question actually is why this doesn't give any Python exception? 
Because it's an error in the C++ API somewhere.

jackbart94, 2020-07-02:
Let me know when it's done, hopefully it was enough to solve this.

Also regarding 0.9.9.3, it's still worth keeping it, even though we hadn't seen this error yet.

zhejun-zhang, 2020-07-03:
I encounters the error `$ what(): length > 2.0f * std::numeric_limits::epsilon()` as well from time to time on 0.9.9 release. I quickly checked the relevant code and it seems to be used to computer the forward_vector, as a sanity check for zero dividing before making a unit vector. I'm not sure if that is the cause of `Segmentation fault (core dumped)`, but instead of sanity check an addition of `eps` seems to be more sensible in this case.

p-testolina, 2020-07-21:
Hi all,
I encountered a similar issue on both 0.9.9 and on the _latest_ docker image. I'm running long simulations to collect data from several sensors. To give you an idea, here you have some of the settings:
- FPS: 60 
- Synchronous mode
- 4 RGB cameras
- 3 LiDARs
- 1 segmentation camera
- Duration (simulation time): 60 seconds
- Simulation runtime: ~1h30min on an NVIDIA 2080 Ti 
This is for a single _scene_, which consists of 3600 frames. 
As I am trying to get diverse scenes, I loop over the available maps, changing the map with client.load_world() while keeping the same Carla server running on the background and the same traffic manager and Carla client.
After a while (it's not deterministic as far as I have observed) I get a `Segmentation fault      (core dumped)`. 

I am **not** getting the `$ what(): length > 2.0f * std::numeric_limits::epsilon()` error log so I'm not sure it's the same issue. It seems to be linked with lengthy simulations (possibly something to do with the frame ID which keeps increasing? - would be nice to have a way to reset the server from the client-side). Please let me know if you want me to open another issue, or if I missed a more relevant one. 

I can provide you additional information or a sample script in the next days if needed. (I have very limited C++ knowledge)

Thanks for the amazing work!

jackbart94, 2020-07-21:
> Hi all,
> I encountered a similar issue on both 0.9.9 and on the _latest_ docker image. I'm running long simulations to collect data from several sensors. To give you an idea, here you have some of [...]

Hi,

it's probably not the same issue, but I encourage you to either open a new GitHub issue with more details, or directly contact us on the CARLA Discord server, so that we can look a bit more into the problem!

germanros1987, 2020-07-22:
@marcgpuig I believe we have encounter the " what(): length > 2.0f * std::numeric_limits::epsilon()" issue before. Do you have anything to add?

KasraMokhtari2112, 2020-08-13:
Hi, I am also doing RL on Carla and have been experiencing the same issue. I am running the same scripts all the time and sometimes it crashes after 5-6 hours, sometimes after 10 hours and so on!! I have no idea why.

I tried to restart Carla every 100-200 episodes which is like almost 1 hour of running but after doing this several times, the simulation crashes with a segmentation error. Therefore, I can't kill and restart Carla lets say more than 10 times. I have to restart Ubuntu and run it again which is so annoying.  Any ideas would be appreciated at this point. I can send my code here if you guys need it. Thanks!

KasraMokhtari2112, 2020-08-13:
Btw, I mostly used Carla 0.9.8.

marcgpuig, 2020-08-18:
@zhejun-zhang can you link me the line of code of the `length > 2.0f * std::numeric_limits::epsilon()`? I'm not able to find it. Maybe it's Unreal's code?

zhejun-zhang, 2020-08-18:
@marcgpuig I found [this line](https://github.com/carla-simulator/carla/blob/6b1a5db4f6c57079efe1f82f881d42dacc4d28fc/LibCarla/source/carla/geom/Vector3D.h#L55) could be relevant. But since I switched to 0.9.9.4 I haven't seen any crash due to this issue.

marintoro, 2020-08-19:
I still got this issue time to time and I switched to CARLA 0.9.9.4.

jackbart94, 2020-08-26:
Hi,

So yeah

> @marcgpuig I found [this line](https://github.com/carla-simulator/carla/blob/6b1a5db4f6c57079efe1f82f881d42dacc4d28fc/LibCarla/source/carla/geom/Vector3D.h#L55) could be relevant. But since I switched to 0.9.9.4 I haven't seen any crash due to this issue.

this line might be what's causing it, however it's because you should be using **MakeSafeUnitVector**. We'll take a look to see if there is something that still might use the other function.

zhejun-zhang, 2020-08-28:
Hi @jackbart94 ,
I'm not sure if that is exactly where the error is thrown. What I did is just searching for `length > 2.0f * std::numeric_limits<float>::epsilon()` int this repository (GitHub search which you find on the top left corner).
And regarding usage. I use only PythonAPI and I don't know which function call leads to this error

closed 2020-09-17 by bernatx

gaoyinfeng, 2021-01-10:
@marintoro , hi! Do you solve this problem? I countered the same problem when training my RL agent with carla 0.9.9(docker).

PR #3243 (2020-09-17): Fixes std::numeric_limits<float>::epsilon error
#### Description

Removes usage of ``MakeUnitVector`` by using ``MakeSafeUnitVector`` and fixes the std::numeric_limits<float>::epsilon error

Fixes #3012

#### Where has this been tested?

  * **Platform(s):** Ubuntu 18
  * **Python version(s):** 3.6
  * **Unreal Engine version(s):** 4.24

#### Possible Drawbacks

None

---
This change is [[image]](https://reviewable.io/reviews/carla-simulator/carla/3243)