carla-simulator/carla #3014: Can not synchronize the vehicle light with street light

little9999, 2020-07-01:
Hi Calra team,

I'm using the calra 0.9.9 on Ubuntu 18.04.
Today I wanna to control automatically the vehicle light in function of the weather. So I have one server active, and then I run "dynamic_weather.py" to simulate the weather changes. THen I modifed the "spawn_npc.py" to check the street light status, and turn on/off the vehicle light.

Here is the code that I added :

` while True:
            # Get the light manager
            light_manager = world.get_lightmanager()
            street_lights = light_manager.get_all_lights(carla.LightGroup.Street)
            light_is_on = light_manager.is_active(street_lights)

            if light_is_on[0]:
                for vehicle_id in vehicles_list:
                    world.get_actor(vehicle_id).set_light_state(carla.VehicleLightState.All)
            else:
                for vehicle_id in vehicles_list:
                    world.get_actor(vehicle_id).set_light_state(carla.VehicleLightState.NONE)

            if args.sync and synchronous_master:
                world.tick()
            else:
                world.wait_for_tick()
`

Visually i can observe the street light turns on / off, however the light_manager.is_active(street_lights)) return always the same value, dose not updated.

Please check it.

germanros1987, 2020-07-02:
@XGodina could you please have a look at this?

marcgpuig, 2020-07-06:
Hi @little9999 sorry but I'm not able to reproduce it, the function `light_manager.is_active()` works fine for me and returns what I expect, maybe you are testing on a town where the `light_is_on[0]` is not set to "on" for any reason? Instead of checking the street lights, I recommend you to check for the sun altitude since they don't need to be all "on".
Furthermore, I encourage you to use the 
```py
light_manager = world.get_lightmanager()
```
outside the loop.

doterop, 2020-07-06:
Hi @little9999,
Perhaps is a problem on the map, which map are you using? If the lights are not tagged properly the group filter will not work.
How many elements are you getting from the `street_lights = light_manager.get_all_lights(carla.LightGroup.Street)`call?

little9999, 2020-07-07:
hi @marcgpuig @doterop 

I've tested with Town01, Town04, Town10HD.
None of them switch the light_manager.is_active(street_lights) status.

I've test with numpy.unique instead of light_is_on[0], all the street lights are off.

And for your proposition of  sun altitude, @marcgpuig . It works well. 
Thanks for your time.

closed 2020-09-12 by stale[bot]