carla-simulator/carla #3016: Ego vehicle at intersection

HarnarayanSingh, 2020-07-02:
Carla Version:0.9.9
OS- Ubuntu 18.04

Is there a way to make ego behavior deterministic at intersection? Like making sure the ego vehicle always takes left turn at a particular intersection. Can we achieve this by editing Maps in Unreal?

jackbart94, 2020-07-02:
Hi,

Right now there isn't, because it depends on many other things, for example the state of traffic lights, or the surrounding traffic. However we are working on determinism for Traffic Manager, so that could come in the near future. Also maps don't really influence the ego vehicle behavior (if you're using ``automatic_control.py`` for example).

HarnarayanSingh, 2020-07-02:
@jackbart94  I found something mentioned in the issue #646 regarding "The routes are picked randomly using the RandomEngine based on the probabilities that have been set for the routes. (These probabilities can be changed in the editor or in-game.)"

So i wanted to know about this possibility. And I'm using manual_control.py.

jackbart94, 2020-07-02:
CARLA has changed a lot in the last two years, that isn't the case anymore. Also if you're using manual_control.py and activating the autopilot, you're using Traffic Manager, so maybe I can give you some further informations in the following days.

HarnarayanSingh, 2020-07-02:
@jackbart94 Thanks .

Bolt0412, 2020-08-05:
Hello, I have the same requriements about the intersection scenario. My CARLA is 0.9.9 with Ubuntu 16.04. Is there any progress on this thread about regulating autopilot vehicle behavior at intersection using traffic manager?

closed 2020-10-11 by stale[bot]