carla-simulator/carla #3019: Filter out Occluded actors using semantic segmentation

Vin1291, 2020-07-02:
Hello, I am trying to filter out the occluded objects using semantic segmentation, but I am not that much familiar with the topic, can anyone help me with it and give me an idea how can I do this in the Carla. I have been through existing topics on this, but not able to figure out the solution. Thank you.

germanros1987, 2020-07-02:
Hi, what are you trying to do exactly?

Vin1291, 2020-07-02:
I am trying to figure out how can I filter out the objects, and I have read old comments of using semantic segmentation or depth, I am not sure which approach would be better and how to actually use it. Can you guide me through it? I would highly appreciate that.

Vin1291, 2020-07-03:
@marcgpuig I have read your comments on using depth on previous issues, can you assist me on this?

marcgpuig, 2020-07-06:
Hi @Vin1291 
Sorry, but since I haven't tried myself I don't think I can guide you more than what I've already posted here on Github that should be enough to implement it and test if it works.
However, I can try to solve more specific doubts on this topic if you face them.

Vin1291, 2020-07-06:
@marcgpuig I have already got 2D bounding box for vehicles, and now I have min_x, min_y, max_x, max_y, but I am not sure how to move further now. Can you tell me how to do the comparison between distance to the actor with depth to the actor.

dev015, 2020-07-10:
@Vin1291 I solved the problem using a depth camera if you are still looking for solution, I can help you with this.

gsujansai, 2020-07-28:
Hey @dev015. I am working on something similar [3028](https://github.com/carla-simulator/carla/issues/3028). Can you share your solution.

dev015, 2020-07-29:
@gsujansai Hey, I have solved it via depth, not with segmentation, are you looking to solve it via depth as well?

gsujansai, 2020-07-29:
@dev015 I am trying to filter out actors that are in front of the ego car or sensor. So only actors that are currently rendered. Although your solution might not be directly related, since you are trying to filter out occluded actors, it might give me some ideas on how to approach my problem. So depth should also be fine.

marcgpuig, 2020-08-18:
@Vin1291 you must use the _**3d pixel**_ from the depth camera and the **3d bounding box** to check if the pixel is **in** it.

closed 2020-11-08 by stale[bot]