carla-simulator/carla #3020: make launch error: wait forever or LogInit: Error:  how to setup  Two TCP ports

gengtsh, 2020-07-02:
hi,
i encounter two extreme cases:
my environment is:
server: ubuntu 20.0.04, use opengl 
remote access from windows 10 using xshell 

1. when no TCP port are set, see the figure
[image]

i got the error: 
[2020.07.02-17.51.17:432][  0]LogInit: Warning: Could not initialize SDL: XRandR support is required but not available
[2020.07.02-17.51.17:432][  0]LogInit:  - Physical RAM available (not considering process quota): 12 GB (11978 MB, 12265568 KB, 12559941632 bytes)
[2020.07.02-17.51.17:433][  0]LogInit:  - VirtualMemoryAllocator pools will grow at scale 1.4
[2020.07.02-17.51.17:433][  0]LogInit:  - MemoryRangeDecommit() will be a no-op (re-run with -vmapoolevict to change)
[2020.07.02-17.51.18:320][  0]LogInit: Using OS detected language (en-US).
[2020.07.02-17.51.18:320][  0]LogInit: Using OS detected locale (en-US).
[2020.07.02-17.51.18:320][  0]LogTextLocalizationManager: No specific localization for 'en-US' exists, so the 'en' localization will be used.
[2020.07.02-17.51.18:496][  0]LogInit: Initializing SDL.
[2020.07.02-17.51.19:452][  0]LogInit: Warning: Could not initialize SDL: XRandR support is required but not available
[2020.07.02-17.51.19:452][  0]LogInit: Warning: FLinuxSplashState::InitSplashResources() : InitSDL() failed, there will be no splash.
[2020.07.02-17.51.19:452][  0]LogInit: Initializing SDL.
[2020.07.02-17.51.21:207][  0]LogInit: Warning: Could not initialize SDL: XRandR support is required but not available
[2020.07.02-17.51.21:207][  0]LogInit: Error: FLinuxApplication::CreateLinuxApplication() : InitSDL() failed, cannot create application instance.
[2020.07.02-17.51.21:207][  0]LogCore: FUnixPlatformMisc::RequestExit(bForce=true, ReturnCode=1)
[2020.07.02-17.51.21:207][  0]LogCore: FUnixPlatformMisc::RequestExit(1)
make: *** [Util/BuildTools/Linux.mk:13: launch] Error 1

2. when i configure the TCP/IP forwarding, see the figure:

[image]

i got the same errors with 1

3. when i configure the TCP/IP in another way, see the figure:
[image]

no window popup,but it looks like then unrealengine is running, but it will be stuck in some place for more then 5 hours.

can someone help me out from this situation? thanks in advance.

germanros1987, 2020-07-02:
@bernatx could you please take a look?

gengtsh, 2020-07-02:
the following are my screen shot , it almost have stay in the same for more than 4 hours:

[2020.07.02-19.01.50:818][  0]Warning:  applied to a non-external Texture2D sampler. This may work by chance on some platforms but is not portable. Please change samer 'EmissiveMap' (slot 1) in material 'M_BuildingMaterial'
[2020.07.02-19.01.50:831][  0]LogEditorServer: Finished looking for orphan Actors (0.000 secs)
[2020.07.02-19.01.51:108][  0]LogUObjectHash: Compacting FUObjectHashTables data took   8.31ms
[2020.07.02-19.01.51:115][  0]Cmd: MAP CHECKDEP NOCLEARLOG
[2020.07.02-19.01.51:116][  0]MapCheck: Map check complete: 0 Error(s), 0 Warning(s), took 0.547ms to complete.
[2020.07.02-19.01.51:116][  0]LogFileHelpers: Loading map 'Town03' took 11.426
[2020.07.02-19.01.51:124][  0]LogLinux: Could not get DPI information for monitor #0, assuming 1.0f
[2020.07.02-19.01.51:124][  0]LogLinux: Could not get DPI information for monitor #0, assuming 1.0f
[2020.07.02-19.01.51:124][  0]LogLinux: Could not get DPI information for monitor #0, assuming 1.0f
[2020.07.02-19.01.51:175][  0]LogSlate: Updating window title bar state: overlay mode, drag disabled, window buttons hidden, title bar hidden
[2020.07.02-19.01.51:184][  0]LogSlate: Updating window title bar state: overlay mode, drag disabled, window buttons hidden, title bar hidden
[2020.07.02-19.01.51:193][  0]LogSlate: Updating window title bar state: overlay mode, drag disabled, window buttons hidden, title bar hidden
[2020.07.02-19.01.51:202][  0]LogSlate: Updating window title bar state: overlay mode, drag disabled, window buttons hidden, title bar hidden
[2020.07.02-19.01.51:218][  0]LogCollectionManager: Loaded 0 collections in 0.002522 seconds
[2020.07.02-19.01.51:230][  0]LogFileCache: Scanning file cache for directory '/home/tgeng/work/carla/Unreal/CarlaUE4/Saved/Collections/' took 0.00s
[2020.07.02-19.01.51:230][  0]LogFileCache: Scanning file cache for directory '/home/tgeng/work/carla/Unreal/CarlaUE4/Content/Developers/tgeng/Collections/' took 0.0
[2020.07.02-19.01.51:230][  0]LogFileCache: Scanning file cache for directory '/home/tgeng/work/carla/Unreal/CarlaUE4/Content/Collections/' took 0.00s
[2020.07.02-19.01.51:230][  0]LogContentBrowser: Native class hierarchy populated in 0.0090 seconds. Added 3025 classes and 670 folders.
[2020.07.02-19.01.52:179][  0]LogContentBrowser: Native class hierarchy updated for 'WidgetCarousel' in 0.0003 seconds. Added 0 classes and 0 folders.
[2020.07.02-19.01.52:205][  0]LogContentBrowser: Native class hierarchy updated for 'AddContentDialog' in 0.0003 seconds. Added 0 classes and 0 folders.
[2020.07.02-19.01.52:607][  0]LogContentBrowser: Native class hierarchy updated for 'SceneOutliner' in 0.0002 seconds. Added 2 classes and 0 folders.
[2020.07.02-19.01.52:671][  0]LogLinux: Could not get DPI information for monitor #0, assuming 1.0f
[2020.07.02-19.01.52:672][  0]LogLinux: Could not get DPI information for monitor #0, assuming 1.0f
[2020.07.02-19.01.52:684][  0]LogLinux: Could not get DPI information for monitor #0, assuming 1.0f
[2020.07.02-19.01.52:684][  0]LogSlate: Window 'Tutorials' being destroyed
[2020.07.02-19.01.52:690][  0]LogLinux: Could not get DPI information for monitor #0, assuming 1.0f
[2020.07.02-19.01.52:690][  0]LogLinux: Could not get DPI information for monitor #0, assuming 1.0f
[2020.07.02-19.01.53:135][  0]LogContentBrowser: Native class hierarchy updated for 'HierarchicalLODOutliner' in 0.0003 seconds. Added 0 classes and 0 folders.
[2020.07.02-19.01.53:135][  0]LogLoad: (Engine Initialization) Total time: 139.53 seconds
[2020.07.02-19.01.53:136][  0]LogLoad: (Engine Initialization) Total Blueprint compile time: 0.00 seconds
[2020.07.02-19.01.53:145][  0]LogRHI: Using OpenGL program LRU cache: 0
[2020.07.02-19.01.53:145][  0]LogSlate: Took 0.000128 seconds to synchronously load lazily loaded font '../../../Engine/Content/Slate/Fonts/Roboto-Regular.ttf' (155K
[2020.07.02-19.01.53:149][  0]LogSlate: Took 0.000105 seconds to synchronously load lazily loaded font '../../../Engine/Content/Slate/Fonts/Roboto-Bold.ttf' (160K)
[2020.07.02-19.01.53:156][  0]LogSlate: Took 0.000090 seconds to synchronously load lazily loaded font '../../../Engine/Content/Editor/Slate/Fonts/FontAwesome.ttf' (
[2020.07.02-19.01.53:988][  0]LogRenderer: Reallocating scene render targets to support 632x360 Format 10 NumSamples 1 (Frame:1).
[2020.07.02-19.01.53:989][  0]LogContentStreaming: Texture pool size now 4000 MB
[2020.07.02-19.01.54:168][  0]LogSlate: Took 0.000081 seconds to synchronously load lazily loaded font '../../../Engine/Content/Slate/Fonts/Roboto-Regular.ttf' (155K
[2020.07.02-19.01.54:473][  0]LogLinux: Setting swap interval to 'Immediate'
[2020.07.02-19.01.54:473][  0]LogLinux: Warning: Unable to set desired swap interval 'Immediate'
[2020.07.02-19.01.54:473][  0]LogLinux: Setting swap interval to 'Immediate'
[2020.07.02-19.01.54:473][  0]LogLinux: Warning: Unable to set desired swap interval 'Immediate'
[2020.07.02-19.01.54:483][  0]LogSlate: Took 0.000073 seconds to synchronously load lazily loaded font '../../../Engine/Content/Slate/Fonts/Roboto-Light.ttf' (167K)
[2020.07.02-19.01.55:203][  7]LogAssetRegistry: Asset discovery search completed in 120.8791 seconds
[2020.07.02-19.01.55:210][  7]LogCollectionManager: Fixed up redirectors for 0 collections in 0.000004 seconds (updated 0 objects)

bernatx, 2020-07-06:
Hi,
I think the important ones are these:
```
[2020.07.02-17.51.21:207][ 0]LogInit: Warning: Could not initialize SDL: XRandR support is required but not available
[2020.07.02-17.51.21:207][ 0]LogInit: Error: FLinuxApplication::CreateLinuxApplication() : InitSDL() failed, cannot create application instance.
```
I think the SDL library has failed.
Also it seems the server is up and running.

gengtsh, 2020-07-06:
Thanks @bernat,
When I turn off the X11 forwarding, this error will be gone. And after a
while it will be stuck forever.
I also try to install SDL library, but it's failed. Some depends are
missing and and then try to install those missings, others are pop up. Even
I try apt install - f xxx  , it also failed.
Can you please provide some suggestions to solve this problem? Thanks .

closed 2020-09-12 by stale[bot]