carla-simulator/carla #3021: errors when make PythonAPI

gengtsh, 2020-07-02:
when i run: make PythonAPI, i got the following errors:

my python version is 3.8

Setup.sh: llvm-8.0 already installed.
Setup.sh: boost-1.72.0-c8 already installed.
Setup.sh: rpclib-v2.2.1_c2-c8 already installed.
Setup.sh: gtest-1.8.1-c8 already installed.
Setup.sh: recast-cdce4e-c8 already installed.
Setup.sh: CARLA version 0.9.9.4-1-gb695dbf2-dirty.
Setup.sh: Generating CMake configuration files.
Setup.sh: Success!
BuildLibCarla.sh: Building LibCarla "Client.Release" configuration.
ninja: no work to do.
[0/1] Install the project...
-- Install configuration: "Client"
BuildLibCarla.sh: Success!
BuildPythonAPI.sh: Building Python API for Python 2.
compiling:
  - source/libcarla/libcarla.cpp
running bdist_egg
running egg_info
writing source/carla.egg-info/PKG-INFO
writing top-level names to source/carla.egg-info/top_level.txt
writing dependency_links to source/carla.egg-info/dependency_links.txt
reading manifest file 'source/carla.egg-info/SOURCES.txt'
writing manifest file 'source/carla.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-x86_64/egg
running install_lib
running build_py
running build_ext
creating build/bdist.linux-x86_64/egg
creating build/bdist.linux-x86_64/egg/carla
copying build/lib.linux-x86_64-2.7/carla/libcarla.so -> build/bdist.linux-x86_64/egg/carla
copying build/lib.linux-x86_64-2.7/carla/__init__.py -> build/bdist.linux-x86_64/egg/carla
copying build/lib.linux-x86_64-2.7/carla/command.py -> build/bdist.linux-x86_64/egg/carla
byte-compiling build/bdist.linux-x86_64/egg/carla/__init__.py to __init__.pyc
byte-compiling build/bdist.linux-x86_64/egg/carla/command.py to command.pyc
creating stub loader for carla/libcarla.so
byte-compiling build/bdist.linux-x86_64/egg/carla/libcarla.py to libcarla.pyc
creating build/bdist.linux-x86_64/egg/EGG-INFO
copying source/carla.egg-info/PKG-INFO -> build/bdist.linux-x86_64/egg/EGG-INFO
copying source/carla.egg-info/SOURCES.txt -> build/bdist.linux-x86_64/egg/EGG-INFO
copying source/carla.egg-info/dependency_links.txt -> build/bdist.linux-x86_64/egg/EGG-INFO
copying source/carla.egg-info/top_level.txt -> build/bdist.linux-x86_64/egg/EGG-INFO
writing build/bdist.linux-x86_64/egg/EGG-INFO/native_libs.txt
zip_safe flag not set; analyzing archive contents...
creating 'dist/carla-0.9.9-py2.7-linux-x86_64.egg' and adding 'build/bdist.linux-x86_64/egg' to it
removing 'build/bdist.linux-x86_64/egg' (and everything under it)
BuildPythonAPI.sh: Building Python API for Python 3.
Traceback (most recent call last):
  File "setup.py", line 160, in <module>
    ext_modules=get_libcarla_extensions(),
  File "setup.py", line 37, in get_libcarla_extensions
    linux_distro = platform.dist()[0]  # pylint: disable=W1505
AttributeError: module 'platform' has no attribute 'dist'
make: *** [Util/BuildTools/Linux.mk:89: PythonAPI] Error 1

germanros1987, 2020-07-02:
could you please provide more information about your setup>

Also, it seems is building the library for 2.7. @bernatx could you take a look at this, please?

gengtsh, 2020-07-02:
I follow the the steps described https://carla.readthedocs.io/en/latest/build_linux/
when i make launch it was stuck in some place, so i make PythonAPI, then i got error,
i remotely access Ubuntu 20.04 through xshell.
server: 
os: ubuntu
python version: python 3.8.2

i search online, it seems that "platform.linux_distribution which was deprecated in Python 3.5 and removed in Python master branch"
in this case, do i need to uninstall python3.8 and reinstall python3.5? and please let me know what information you need, thanks

bernatx, 2020-07-06:
Hi, we have not tested yet in Ubuntu 20.04, but you can try installing Python x64 3.7.2
It seems that Python 3.8 could be in trouble with boost. We are investigating.
Let us know if that solves your problem, please.
Regards

gengtsh, 2020-07-07:
i install python3.7 , and set in my 
.baserc : 
alias python = python3.7
alias python3 = python2.7

to double check python version, 
when i run python --version and python3 --version, i got Python 3.7.0

and rerun make PythonAPI, 
i got the same error
BuildPythonAPI.sh: Building Python API for Python 3.
Traceback (most recent call last):
  File "setup.py", line 160, in <module>
    ext_modules=get_libcarla_extensions(),
  File "setup.py", line 37, in get_libcarla_extensions
    linux_distro = platform.dist()[0]  # pylint: disable=W1505
AttributeError: module 'platform' has no attribute 'dist'
make: *** [Util/BuildTools/Linux.mk:89: PythonAPI] Error 1

bernatx, 2020-07-13:
With new version of Python some deprecated functionality has been removed finally, and now we are using a new **distro** python module to know which distribution of linux your are running. You could get last version of code and install the module with **pip3 install distro**. It should work.

closed 2020-09-20 by stale[bot]

alexkutsan, 2022-04-28:
Answering my previous question. Python3.7 works fine 
```
╰─○ docker run -it python3.7 /bin/bash
Unable to find image 'python3.7:latest' locally
docker: Error response from daemon: pull access denied for python3.7, repository does not exist or may require 'docker login': denied: requested access to the resource is denied.
See 'docker run --help'.
╭─akutsan at powerpc in ~/rnd/company using «venv»
╰─○ docker run -it python:3.7 /bin/bash
Unable to find image 'python:3.7' locally
3.7: Pulling from library/python
6aefca2dc61d: Pull complete 
967757d56527: Pull complete 
c357e2c68cb3: Pull complete 
c766e27afb21: Pull complete 
32a180f5cf85: Pull complete 
1535e3c1181a: Pull complete 
8ae21bbc5192: Pull complete 
7d9c83d514b0: Pull complete 
52e9a986c056: Pull complete 
Digest: sha256:7160fc73878a5f035e932f0d88f6e606f593580c442270c30765c7768bc7bc3f
Status: Downloaded newer image for python:3.7
root@e25fb3a45210:/# pip3 install carla
Collecting carla
  Downloading carla-0.9.13-cp37-cp37m-manylinux_2_27_x86_64.whl (29.7 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 29.7/29.7 MB 911.4 kB/s eta 0:00:00
Installing collected packages: carla
Successfully installed carla-0.9.13
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
```

Med-srata, 2023-07-16:
Hello guys! 

please I found the same problem ! 
Traceback (most recent call last):
  File "c:/Users/PC/Desktop/WindowsNoEditor/PythonAPI/examples/test.py", line 1219, in <module>
    main()
  File "c:/Users/PC/Desktop/WindowsNoEditor/PythonAPI/examples/test.py", line 1212, in main
    game_loop(args)
  File "c:/Users/PC/Desktop/WindowsNoEditor/PythonAPI/examples/test.py", line 1123, in game_loop
    world = World(client.load_world('Town04'), hud, args.filter,
RuntimeError: time-out of 2000ms while waiting for the simulator, make sure the simulator is ready and connected to 127.0.0.1:2000

how can I solve it ?