carla-simulator/carla #3022: Is it possible to drive on the left with CARLA 0.9.9?

shishau, 2020-07-03:
Hello everyone,

When I import the road data of the left-hand side created by RoadRunner into Carla 0.9.9, it becomes the road of the right-hand side without complying with the RoadRunner settings.

When I contacted RoadRunner support, I was told that it was Carla 0.9.9 specification.
Is this true?
Isn't Carla 0.9.9 possible to simulate left-hand traffic like Japan and the UK?

Please let me know if anyone knows.
Thank you.

shishau

germanros1987, 2020-07-06:
Hi@shishau, 

CARLA doesn't support left-side traffic yet. This feature is planned for future releases.

shishau, 2020-07-07:
Hi. @germanros1987

Thank you for your reply.
I look forward to future releases.

shishau

closed 2020-07-30 by germanros1987

joealan11, 2020-09-10:
Any advice to where to look for the lef hand traffic to work? We were doing it with route planner, but now this is deprecated.

bg-sumu, 2024-08-17:
Any progress on the same? Could we use Left hand Driving in Carla 0.9.15 ?

JIANGZhengm, 2024-09-11:
> Hi@shishau,
> 
> CARLA doesn't support left-side traffic yet. This feature is planned for future releases.
hello, I am using CARLA 0.9.15，Is it now possible to drive on the left?