carla-simulator/carla #3023: Option to toggle visibility of lane-lines

bender2po, 2020-07-03:
Hi, is there any option to switch off (or deteriorate) visibility of the lane markings in Carla. The purpose is to train the CNN model for detection of lane-markings even in the absence of lane-markings. So, I would be needing images from the camera without lane-markings. Let me know if there is any work around if not the availability of visibility-switch.

germanros1987, 2020-07-06:
Hi @bender2po,

This is not possible directly from the Python API, but you could modify the existing UE4 tools to achieve this. Have a look at the tools presented here:

https://www.youtube.com/watch?v=ZZaHevsz8W8&feature=youtu.be
https://drive.google.com/file/d/1l9Ztaq0Q8fNN5YPU4-5vL13eZUwsQl5P/view

Pay attention to the BP Paint Road. That should be a good starting point for your problem.

bender2po, 2020-07-07:
@germanros1987 Thank you for the reply.
I can proceed with your suggestion now.

closed 2020-07-11 by bender2po