carla-simulator/carla #3025: More detailed documents of LibCarla

BigTailFox, 2020-07-04:
Hello,
I am using LibCarla to create a C++ client, however in the LibCarla documents little information was given, such as the units of variables, the world/vehicle/sensor coordinate systems. Is there any plan of a more detailed LibCarla document?

germanros1987, 2020-07-06:
@BigTailFox tanks for your feedback

We are continuously working on improving the documentation. I am adding @sergi-e to this conversation so that he can hear directly from you about documentation improvements. Could you list the things that you would like to see improved?

Thanks.

BigTailFox, 2020-07-09:
Thank you, below list my suggestions:
1. parameters' and arguments' descriptions in LibCarla, including units, coordinate systems.
2. detailed descriptions of some methods, such as `getID` and `getTypeID` in `carla::client::Actor`，`getID` in `ActorBlueprint`, a little bit confusion.

by the way, I compiled LibCarla of 0.9.9.4 version, and I want to get the bounding box of an actor, but `Actor` has a protected method `GetBoundingBox`, thus I can't access to it. In PythonAPI an `Actor` object is able to access it's bounding box directly by attribute `bbox`.

sergi-e, 2020-07-09:
@BigTailFox  Thanks a lot for the suggestions! I will update this as soon as possible.

closed 2020-09-20 by stale[bot]