carla-simulator/carla #3027: Collision sensor not working with syncronous mode and fixed time step

ahmedhatem3h, 2020-07-04:
I have created a scenario where there are 2 vehicles colliding and I can't get the collision sensor attached to the vehicle to sense the collision between them. Bellow is the code I run and callback doesn't get called at all.

`
import glob
import os
import sys
import time
import numpy as np

try:
    sys.path.append(glob.glob('../carla/dist/carla-*%d.%d-%s.egg' % (
        sys.version_info.major,
        sys.version_info.minor,
        'win-amd64' if os.name == 'nt' else 'linux-x86_64'))[0])
except IndexError:
    pass
import carla

class Mavim(object):
    
    def __init__(self):

        self.actor_list = []

        try:
            self.client = carla.Client('localhost', 2000)
            self.client.set_timeout(2.0)

            self.world = self.client.get_world()

            settings = self.world.get_settings()
            settings.fixed_delta_seconds = 0.05
            settings.synchronous_mode = True
            self.world.apply_settings(settings)

            self.world.tick()
            self.map = self.world.get_map()

            self.blueprint_library = self.world.get_blueprint_library()
            
            self.vehicle = None
            self.collision_sensor = None
            self.spawn_actor()

            self.world.tick()

            while True:
                self.vehicle.apply_control(carla.VehicleControl(throttle = 1 , brake = 0, steer = 0))
                self.world.tick()
                time.sleep(0.05)

        finally:
            settings = self.world.get_settings()
            settings.fixed_delta_seconds = None
            settings.synchronous_mode = False
            self.world.apply_settings(settings) 

            print('destroying actors')
            for actor in self.actor_list:
                actor.destroy()
            
            print('done.')
    
    def collision_data(self, event):
        print("collided")
    
    def spawn_actor(self):
        
        bp = self.blueprint_library.filter('model3')[0]
        print(len(self.blueprint_library.filter('model3')))
        spawn_point = carla.Transform(carla.Location(-100, 2.8, 2),carla.Rotation(yaw=0))
        self.vehicle = self.world.spawn_actor(bp, spawn_point)
        self.actor_list.append(self.vehicle)

        spawn_point2 = carla.Transform(carla.Location(-55, 2.8, 2),carla.Rotation(yaw=90))
        vehicle2 = self.world.spawn_actor(bp, spawn_point2)
        self.actor_list.append(vehicle2)

        colsensorbp = self.blueprint_library.find('sensor.other.collision')

        transform = carla.Transform()
        self.collision_sensor = self.world.spawn_actor(colsensorbp, transform, attach_to=self.vehicle)
        self.actor_list.append(self.collision_sensor)

        self.collision_sensor.listen(lambda event: self.collision_data(event))

def main():

    test = Mavim()

if __name__ == '__main__':

    main()
`

I am using:
CARLA 0.9.7
Ubuntu 18.04
Python 2.7.17

Please help me as I am stuck at this for couple of days already and I can't get it to work.

ahmedhatem3h, 2020-07-04:
Edit
Actually I turned off synchronous mode and it still doesn't work

ahmedhatem3h, 2020-07-04:
I simplified the code as much as possible and still not working
`import glob
import os
import sys
import time
import numpy as np
import queue

try:
    sys.path.append(glob.glob('../carla/dist/carla-*%d.%d-%s.egg' % (
        sys.version_info.major,
        sys.version_info.minor,
        'win-amd64' if os.name == 'nt' else 'linux-x86_64'))[0])
except IndexError:
    pass
import carla

actor_list = []

client = carla.Client('localhost', 2000)
client.set_timeout(2.0)

world = client.get_world()

collision_q = []
map = world.get_map()

blueprint_library = world.get_blueprint_library()

collision_hist = []

bp = blueprint_library.filter('model3')[0]
print(len(blueprint_library.filter('model3')))
spawn_point = carla.Transform(carla.Location(-100, 2.8, 2),carla.Rotation(yaw=0))
vehicle = world.spawn_actor(bp, spawn_point)
actor_list.append(vehicle)

spawn_point2 = carla.Transform(carla.Location(-55, 2.8, 2),carla.Rotation(yaw=90))
vehicle2 = world.spawn_actor(bp, spawn_point2)
actor_list.append(vehicle2)

colsensor = world.spawn_actor(blueprint_library.find('sensor.other.collision'),
                                    carla.Transform(), attach_to=vehicle)
actor_list.append(colsensor)
colsensor.listen(lambda event: collision_data(event))

print(colsensor.is_listening)
vehicle.apply_control(carla.VehicleControl(throttle = 1 , brake = 0, steer = 0))

while True:
    time.sleep(0.05)

print('destroying actors')
for actor in actor_list:
    actor.destroy()

print('done.')

def collision_data(event):
        print("collided")`

AlexScheid, 2020-07-06:
Hello @ahmedhatem3h,
I tested the last provided code with CARLA 0.9.9. If you put the `collision_data` function somewhere in front of the callback definition at `colsensor.listen(lambda event: collision_data(event))` the code is working fine.
Also the definition of the collision event in the class above is working.
So I think it is a problem in version 0.9.7

ahmedhatem3h, 2020-07-06:
Hello @AlexScheid,
Thank you. Yes It is working fine with CARLA 0.9.9.4. I have read the changelog files and there was no mention of an update in the collision sensor, However, there is a difference between CARLA 0.9.7 and CARLA 0.9.9.4 so hope someone add this so people can know that it is not working in CARLA 0.9.7

closed 2020-07-06 by ahmedhatem3h

qhaas, 2020-09-29:
Unfortunately, our workflow can't handle the version of UnrealEngine that CARLA 0.9.9.x was built with, so we were using CARLA 0.9.7, until we ran into this bug...  It doesn't appear to occur in CARLA 0.9.6, and we haven't tested CARLA 0.9.8 yet.

Was the cause/fix ever narrowed down so we can backport the fix?

Thanks