carla-simulator/carla #3028: How to get actors that are currently rendered

gsujansai, 2020-07-04:
Hi

i would like to get a list of all actors that are currently rendered in scene, so that i know which actors in world are being captured by the camera sensor or the segmentation sensor at that instant. I found that unreal engine has the function was [WasRecentlyRendered](https://docs.unrealengine.com/en-US/API/Runtime/Engine/GameFramework/AActor/WasRecentlyRendered/index.html). This doesn't seem to be available in the Python API. So can someone suggest how I can solve this problem. 

Note: already asked this question on carla forum but that doesnt seem to be active so I m posting here.
https://forum.carla.org/t/get-walkers-information-if-they-are-in-sensor-view/326

xmyqsh, 2020-07-04:
[WasRecentlyRendered](https://docs.unrealengine.com/en-US/API/Runtime/Engine/GameFramework/AActor/WasRecentlyRendered/index.html) is a method of Actor. You can expand it as an actor attribute in Python API. Or, add it as a special filter into ActorList which should be lowest the burden of RPC and match your demand.
```
  class_<cc::ActorList, boost::shared_ptr<cc::ActorList>>("ActorList", no_init)
    .def("find", &cc::ActorList::Find, (arg("id")))
    .def("filter", &cc::ActorList::Filter, (arg("wildcard_pattern")))
    .def("__getitem__", &cc::ActorList::at)
    .def("__len__", &cc::ActorList::size)
    .def("__iter__", range(&cc::ActorList::begin, &cc::ActorList::end))
    .def(self_ns::str(self_ns::self))
  ;
```

gsujansai, 2020-07-07:
Thanks @xmyqsh . I don't have much experience working with the cpp codebase of carla. I will try it out and post my status. 

As a quick hack, i used carla transforms with some offset in front of the camera sensor. I look for all pedestrians transforms that are inside this offset point and camera center region. Ofcourse, the drawback is that actors that are behind buildings or occlusions are also considered as inCameraView using this quick hack.  

@marcgpuig I m open to any other approach to solving this problem.

gsujansai, 2020-08-06:
Hi @xmyqsh @marcgpuig . I have created a filter function in ActorList like you suggested. But i did it for the C++ API first. 

`SharedPtr<ActorList> ActorList::NewFilter(const std::string &wildcard_pattern) const {
    SharedPtr<ActorList> filtered (new ActorList(_episode, {}));
    for (auto &&actor : _actors) {      
        if (actor->WasRecentlyRendered()) {
            filtered->_actors.push_back(actor);
       }
}` 

This change is made in [Actorlist.cpp](https://github.com/carla-simulator/carla/blob/master/LibCarla/source/carla/client/ActorList.cpp)
Unfortunately WasRecentlyRendered is not exposed to the CARLA_API. Can you give me some hints on where i should add the method. I assume its in some actor attributes file [Unreal/Plugin](https://github.com/carla-simulator/carla/tree/master/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Actor) but I havent had success with my attempts until now.

closed 2020-10-18 by stale[bot]