carla-simulator/carla #3029: Default port 2000 unusable on Windows 10 with Hyper-V

lukase, 2020-07-04:
When running carla on a Windows 10 machine with latest updates and Hyper-V, it crashes due to an unhandled exception while trying to bind to port 2000 (TCP). This happens with release binaries as well as when running from the Unreal editor.

Port 2000 is reserved by Hyper-V, and cannot be used by other programs. The behaviour was introduced in Windows 10 update [KB4074588](https://support.microsoft.com/en-us/help/4074588/windows-10-update-kb4074588)

A temporary workaround can be found [here](https://stackoverflow.com/questions/48478869/cannot-bind-to-some-ports-due-to-permission-denied/54727281#54727281)

Would it be possible to change the default port to lie outside the reserved ranges?

germanros1987, 2020-07-06:
Why don't you launch it in a different port?

closed 2020-08-29 by germanros1987

badrow, 2024-03-28:
I don't even now what is Carla but I encounter a similar problem while listening on port 2000 on a linux machine running on hyper-v.

Port 2000 is not working outside the VM. Really weird hyper-v bug.