carla-simulator/carla #3030: issues for ./UE4Editor--vulkan

gengtsh, 2020-07-04:
hi,
i encounter some issue  about ./UE4Editor, since vulkan is necessary for unrealengine 4.24, i install vulkan on the server
i use xshell (on windows 10) remote access ubuntu 18.04 server.
windows has 3 monitors
Ubuntu server have two nvidia Quadro P6000 cards
vulkaninfo: version: 1.2.141.2

  1. xshell: forward X11 connections to Xmanager
    when i run vulkaninfo, i got one line:
    ERROR at /root/sdk-build/1.2.141.2/source/Vulkan-Tools/vulkaninfo/vulkaninfo.h:240:vkEnumerateInstanceExtensionProperties failed with ERROR_INITIALIZATION_FAILED

    2. xshell : forward X11 X DISPLAY localhost:0.0
    when i run vulkaninfo, i get two lines:
    XCB failed to connect to the X server due to error:1.
    ERROR at /root/sdk-build/1.2.141.2/source/Vulkan-Tools/vulkaninfo/vulkaninfo.h:821: AppCreateXcbSurface failed to establish connection

3. when in turn off X11 Forward in xshell, login to ubuntu server though xshell and  run ./UE4Editor
i got: LogLinux: Warning: MessageBox: Failed to load Vulkan Driver which is required to run the engine.
The engine no longer fallbacks to OpenGL4 which has been deprecated.: Message:

when i run vulkaninfo, i can get some information in shell, but have some issues to run othe vulken tools. 
such as : when i run vkconfig & , i got the following errors:
QXcbConnection: Failed to initialize XRandr
libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: swrast

i have struggled for two days, any help are welcome. thanks a lot

germanros1987, 2020-07-06:
@gengtsh thanks for reporting your problem.

Let me try to make sense out of it since your description is a bit confusing. It seems you are describing a setup involving 2 machines? A Ubuntu 18.04 server and a Windows 10 node. 

Is the Ubuntu 18.04 running the server? Is the Windows 10 node running the client? Is that what you are trying to do?

P

gengtsh, 2020-07-06:
Hi, germanros,
Yes, there are two machines, Ubuntu 18.04 server and windows 10 as client
,  which remotely access Ubuntu.

germanros1987, 2020-07-30:
@gengtsh this has to do with the way the xserver is configured. A few points:

- Have you tried running the docker image of CARLA in the server?
- Do you really need X forwarding? Could you just set up a camera to receive images on the client?

gengtsh, 2020-08-03:
thanks for your help @germanros1987.
i don't really need X forwarding, when i connection on monitor with ubuntu server and use teamviewer remotely access ubuntu and then i can run smoothly. 
and i will run docker image also in next step. thanks again.

closed 2020-08-29 by germanros1987