carla-simulator/carla #3033: Following a vehicle

ayushshah, 2020-07-06:
I am a beginner trying to learn Carla. Thank you for the detailed documentation, but I am unable to find how to achieve the following. 

I wish to create a simulation in which vehicle_1 follows the path being followed by vehicle_2 while keeping the throttle and brake control independent of the movement of vehicle_2. 

I tried specifying it while spawning vehicle_1 as follows, but that does not work.
`vehicle_1 = try_spawn_actor(blueprint, spawn_point, attach_to=self.vehicle_2, attachment_type=carla.AttachmentType.Rigid)`

Can someone please give a pointer on how this can be done?

GabiShahmayster, 2020-07-08:
@ayushshah 
Hi,

Are you sure this can be done? 
Because this type of attachment seems more suitable for a sensor, than a vehicle... 
If this was a real-life scenario, the simplest solution, which comes to mind, would be for the following vehicle to apply controls, in such a way that the a certain distance between it and the leading vehicle would be maintained. 
You would build a controller, which would be driven by the distance to the lead vehicle. 
As a start, you could try to use the VehiclePIDController (from PythonAPI/agents/navigation/controller.py), continuously updating it's target waypoint to a lead vehicle's past location. 
A better approach would be retrieve the distance to the lead vehicle, using the lead vehicle's and the following vehicle's locations and feed it to a controller.

Good luck,
Gabi Shahmayster

ayushshah, 2020-07-08:
Thank you for the detailed answer. I am able to follow the lead vehicle using controller.py. 

Cheers!

closed 2020-07-08 by jackbart94