carla-simulator/carla #3035: sensor_tick property ignored for GNSS, Radar and IMU

Kipt, 2020-07-07:
After troubleshooting why sensor_tick for IMU didn't change the frequency of the ROS message I noticed the ego_vehicle.py doesn't read the property. This is true for GNSS, radar and IMU sensors. Adding the 
```
try:
    bp.set_attribute('sensor_tick', str(sensor_spec['sensor_tick']))
except KeyError:
    pass
```
snippet fixes the issue.

germanros1987, 2020-07-07:
@joel-mb @fpasch could you comment on this issue?

Kipt, 2020-09-12:
Is this how bugs are "fixed"?

joel-mb, 2020-09-14:
Hi @Kipt 
Sorry for the delay. I missed this issue. Could you open a PR in the ros-bridge introducing the fix. Otherwise, I'll do it myself.

Kipt, 2020-09-16:
Thanks for checking this out @joel-mb 

As the commit I have is based on a ROS-bridge repository that's heavily modified already I think it's faster if you do the change.

closed 2020-09-24 by fpasch