carla-simulator/carla #3038: How to add license plate on vehicle

Georgelingzj, 2020-07-07:
Hi,

I am currently work on adding license plate on vehicles.

I wonder is it possible to create it in python API or i need to do it in Unreal engine

I have looked through the carla documents and there is nothing about this.

Thanks!

germanros1987, 2020-07-07:
Hi @Georgelingzj you would need to create a new blueprint / modify vehicle's blueprint to do this. We would be happy to take this as a contribution :)

Georgelingzj, 2020-07-08:
> Hi @Georgelingzj you would need to create a new blueprint / modify vehicle's blueprint to do this. We would be happy to take this as a contribution :)

I am trying bit by bit. I would like to build a function that generate random license plate on vehicles. But it seems quite tough.

XGodina, 2020-07-09:
@Georgelingzj We are working on it. As soon as possible  I'll let you know.

closed 2020-09-20 by stale[bot]