carla-simulator/carla #3040: missing module name in carla-0.9.9-py3.6-linux-x86_64.egg

jeffchen006, 2020-07-07:
Carla version: 0.9.9
Platform: Ubuntu 18.04

Problem description: After following the installation instructions in https://carla.readthedocs.io/en/latest/build_linux/, I failed at the last step, i.e. python3 spawn_npc.py

The error messages are listed below:
<pre>Traceback (most recent call last):
  File &quot;spawn_npc.py&quot;, line 27, in &lt;module&gt;
    import carla
  File &quot;&lt;frozen importlib._bootstrap&gt;&quot;, line 971, in _find_and_load
  File &quot;&lt;frozen importlib._bootstrap&gt;&quot;, line 955, in _find_and_load_unlocked
  File &quot;&lt;frozen importlib._bootstrap&gt;&quot;, line 656, in _load_unlocked
  File &quot;&lt;frozen importlib._bootstrap&gt;&quot;, line 626, in _load_backward_compatible
  File &quot;../carla/dist/carla-0.9.9-py3.6-linux-x86_64.egg/carla/__init__.py&quot;, line 8, in &lt;module&gt;
  File &quot;&lt;frozen importlib._bootstrap&gt;&quot;, line 971, in _find_and_load
  File &quot;&lt;frozen importlib._bootstrap&gt;&quot;, line 955, in _find_and_load_unlocked
  File &quot;&lt;frozen importlib._bootstrap&gt;&quot;, line 656, in _load_unlocked
  File &quot;&lt;frozen importlib._bootstrap&gt;&quot;, line 626, in _load_backward_compatible
  File &quot;../carla/dist/carla-0.9.9-py3.6-linux-x86_64.egg/carla/libcarla.py&quot;, line 8, in &lt;module&gt;
  File &quot;../carla/dist/carla-0.9.9-py3.6-linux-x86_64.egg/carla/libcarla.py&quot;, line 7, in __bootstrap__
TypeError: exec_module() missing 1 required positional argument: &apos;module&apos;
</pre>

I looked into the ../carla/dist/carla-0.9.9-py3.6-linux-x86_64.egg/carla/libcarla.py, and the following is its content:
```
def __bootstrap__():
    global __bootstrap__, __loader__, __file__
    import sys, pkg_resources
    from importlib.machinery import ExtensionFileLoader
    __file__ = pkg_resources.resource_filename(__name__, 'libcarla.cpython-36m-x86_64-linux-gnu.so')
    __loader__ = None; del __bootstrap__, __loader__
    ExtensionFileLoader(__name__,__file__).exec_module()
__bootstrap__()
```

I don't knnow how an egg file is generated here but I think there does miss a module name here(in ```exec_module()``` ). But I don't know how to modify it to make PythonAPI work. So I'm asking for some help here. Any suggestion/thoughts are warmly welcomed.

germanros1987, 2020-07-09:
@bernatx could you please comment on this?

Doodle1106, 2020-07-10:
same problem here

jeffchen006, 2020-07-12:
Update: upgrade my Python from 3.6 to 3.7. and relaunch. Get new egg files. Everyting goes fine now.

bernatx, 2020-07-13:
Probably your previous .egg with version 3.6 was incorrect. If you don't want to upgrade Python version try to use **make clean PythonAPI** and see if the new .egg works.

jeffchen006, 2020-07-14:
I'm pretty sure I did `make clean` and `make PythonAPI` previously and I got the error again and again with Python3.6.
I'm no idea how carla generates .egg files, but I'm suspicious there is something wrong with the process to generate .egg files.
Though I solved this problem by upgrading Python and relaunch, it would be great if some collaborators of carla project can take a look at carla-0.9.9 with python3.6 to avoid the similar confusion in the future. Thanks.

closed 2020-07-18 by jeffchen006

yasser-h-khalil, 2020-07-25:
I am facing the same issue as @jeffchen006. With Ubuntu 18.04, Python3.6 comes by default. I am getting .egg of 3.6 which fails to launch the python examples in ~/carla/PythonAPI/examples. I tried upgrading python to 3.7 but failed. 

@jeffchen006 could you provide help with how did you upgrade your python from 3.6 to 3.7. And what are the following steps post python upgrade? 

Your help is appreciated.

HemaZ, 2020-09-07:
a workaround would be renaming the .egg file from  **carla-0.9.9-py3.7-linux-x86_64.egg**  to  **carla-0.9.9-py3.6-linux-x86_64.egg**  . it works for me with no issues