carla-simulator/carla #3044: How to get velocity at center of mass of ego vehicle?

gershom96, 2020-07-09:
I have been trying to get the velocity of the center of mass of my ego vehicle. 

1. I have tried to attach a center to the center of mass of the vehicle and then get its velocity. However, this failed. I assume this is due to the fact that you cannot obtain velocity at a specific point. 

I know you could use an accelerometer sensor and calculate this. However I am looking for a much elegant solution.

germanros1987, 2020-07-14:
Hi @gershom96,

I don't believe the physics engine (PhysX) keeps the speed information at every point. Can't you just take the velocity of the vehicle as a rigid object?

@DSantosO should be able to add more to the topic.

DSantosO, 2020-07-17:
Hello @gershom96,  
Like @germanros1987, I don't think you can recover the velocity at any point but, if all you need is the velocity in the center of mass, is there a reason why the get_velocity() of the actor is not good for you?

closed 2020-09-29 by stale[bot]