carla-simulator/carla #3045: creating an environment with limited borders in CARLA

hnuma, 2020-07-09:
I want to apply Q-learning algorithm in CARLA. I need to create a discrete environment for this, how can I define it on code? I would be very glad if you can help about this.

germanros1987, 2020-07-14:
You should follow the documentation on how to create maps. @sergi-e can provide you with more information.

hnuma, 2020-07-14:
@germanros1987 thank you for your feedback. 

Roadrunner is recommended according to the instructions on the site, but unfortunately there is no download authority. It issued a warning for students.

According to the CARLA site;

Create a map with RoadRunner
Export from RoadRunner
Map ingestion in a CARLA package
Map ingestion in a build from source
Modify pedestrian navigation
Deprecated ways to import a map

If these options are independent of each other, which way should I go? 

I would be very pleased if he could help. @sergi-e

sergi-e, 2020-07-16:
Hi, @hnuma! 
While reading you, a question came to my mind. If you only want to test your agent in simple scenarios such as a roundabout or intersections, is there any reason for you to create a map of your own? 
In CARLA, Town03 has a roundabout and there are intersections in all the maps. Maybe you could use the already existing CARLA maps. 
The ScenarioRunner module will allow you to define scenarios in said maps and create routes to test your agent. 
Here is a link to the ScenarioRunner docs: https://carla-scenariorunner.readthedocs.io/en/latest/
And here is a talk introducing the module: https://www.youtube.com/watch?v=dcnnNJowqzM&t=1s

hnuma, 2020-07-22:
Hi @sergi-e thank you so much information you provided. I followed the paths you said. I downloaded CARLA's and scenario Runner's  last versions. I keep getting certain mistakes. Are there some incompatibilities for windows in CARLA? For example, I get the error "AttributeError: module 'signal' has no attribute 'SIGHUP'". This is mentioned as an error that appears on windows as far as I can see on the internet.

closed 2020-09-29 by stale[bot]