carla-simulator/carla #3046: Can I move a pedestrian via apply_control on Driving road?

AIasd, 2020-07-10:
Hi!

I spawned a pedestrian on a Driving road and then invoked apply_control on it but the pedestrian gradually moved downward (to the ground) and then disappear. I tried to change the direction of the pedestrian to different values but it seems to not have any influence.

germanros1987, 2020-07-14:
@bernatx could you help with this?

@sergi-e could you point to the right documentation?

bernatx, 2020-07-14:
Can you tell us in which map you tested? Is it an standard map included in CARLA or a custom one?

AIasd, 2020-07-14:
Hi!
I actually finally figured it out. It is on Town10HD and I used the scenario runner (I probably should ask there). It seems that the problem is with the execution of 

`
if self._physics:
    self._actor.set_simulate_physics(enabled=True)
` 

in class ActorTransformSetter in atomic_behavior.py. 

After I followed the scenario example (e.g. object_crash_vehicle.py) to set self_physics to False when initializing ActorTransformSetter, the behavior of the pedestrian becomes normal.

Thank you for your responses!

closed 2020-07-27 by AIasd