carla-simulator/carla #3048: Carla compilation environment version and docker Ubuntu version does not match

erdnaxe, 2020-07-10:
`carlamsim/carla` docker image is built from `nvidia/vulkan:1.1.121-cuda-10.1-alpha` which is based on Ubuntu 16.04.
It seems that carla is compiled on Ubuntu 18.04, this causes compatibility issues when this docker image is modified to add Ubuntu's Python 3.5 and libstdc++6.

Why is https://github.com/carla-simulator/carla/blob/master/Util/Docker/Release.Dockerfile based on `nvidia/vulkan:1.1.121-cuda-10.1-alpha` ? Shouldn't it be based rather on `nvidia/vulkan:1.1.121-cuda-10.1--ubuntu18.04` ?

I did my testing with Carla 0.9.9.

Regards,

germanros1987, 2020-07-14:
@erdnaxe this seems to be a good suggestion. We would appreciate a PR with the change.

bernatx, 2020-07-15:
@erdnaxe our docker build in carlasim is build on Ubuntu **16.04**, because is the minimum version we support.
If we apply your change, will it break our docker on Ubuntu 16.04?

erdnaxe, 2020-07-15:
Indeed my change will upgrade carlasim/carla docker to 18.04. Maybe it could be possible to ship two docker images, one based on `nvidia/vulkan:1.1.121-cuda-10.1--ubuntu18.04` and one based on `nvidia/vulkan:1.1.121-cuda-10.1--ubuntu16.04` ?

@bernatx You seem to indicate that Ubuntu 16.04 is used to build carla, but at least in `carlasim/carla:0.9.9` I observed that:
* `/home/carla/PythonAPI/carla/dist/carla-0.9.9-py3.7-linux-x86_64.egg` was built using Python3.7 which is not Ubuntu 16.04 default Python3. This makes it impossible to import in 16.04 Python3 without an additional PPA.
* `/home/carla/Tools/RecastBuilder` was compiled using Ubuntu 18.04 `libstdc++`. This makes it impossible to use without a core dump on 16.04.

That is why I propose to upgrade the docker to match the Ubuntu version. If your goal is to provide an Ubuntu 16.04 docker image, then maybe there is a version mismatch during the building process ?

thytless, 2020-07-20:
I am also confused with this problem. I found your docs about building on linux says 'Since release 0.9.9, CARLA cannot run in Ubuntu 16.04 using default compilers', but release image is based on 16.04 ...

bernatx, 2020-07-20:
The thing is that we support Ubuntu 16.04 for building CARLA, but you need to install the proper version of compilers. The default compilers of Ubuntu 16.04 will not compile CARLA, because UE 4.24 requires newer versions of compilers.

thytless, 2020-07-21:
@bernatx Got it, so where could I find requirements about compiler and libraries to build in Ubuntu 16.04? Thx!

bernatx, 2020-07-21:
In this link: https://carla.readthedocs.io/en/latest/build_linux/
You have requeriments for Ubuntu 18.04 and for previous versions (16.04)

[image]

PR #3054 (2020-09-02): Upgrade carla Docker image to Ubuntu 18.04
Update `carlamsim/carla` base image from `nvidia/vulkan:1.1.121-cuda-10.1-alpha` (Ubuntu 16.04) to `nvidia/vulkan:1.1.121-cuda-10.1--ubuntu18.04` (Ubuntu 18.04).

This enables user to extend this docker image with the right Python and libstdc++6 versions, as carla is built on 18.04.

Fixes #3048

---
This change is [[image]](https://reviewable.io/reviews/carla-simulator/carla/3054)
closed 2020-07-22 by germanros1987