carla-simulator/carla #3050: details about collisions on carla-0.9.9

whuxhzhang, 2020-07-13:
Can Carla record more details about the collision, such as whether my car hit someone else or someone else hit my car

germanros1987, 2020-07-14:
Hi @whuxhzhang 

Our goal in designing the recording system was to provide the minimum information that is needed in order to detect all possible events. However, we didn't plan to record all these events by default. You could parse the log and generate that information to assign responsibility to the right vehicle. 

Hope this helps.

whuxhzhang, 2020-07-14:
> Hi @whuxhzhang
>
> Our goal in designing the recording system was to provide the minimum information that is needed in order to detect all possible events. However, we didn't plan to record all these [...]

Thanks for your replay. But where can I find the log ？

germanros1987, 2020-07-30:
https://carla.readthedocs.io/en/latest/adv_recorder/

closed 2020-08-29 by germanros1987